desbma/hddfancontrol #3: Two drives, one fan: Invalid parameter count

reinersmann, 2015-06-05:
I am not sure whether this is really an issue or if I am not handling things right:
I have installed the latest version of hddfancontrol (first install) and had problems to get it to run. I have two drives that I want to cool down with one fan. So I took your example and removed the second fan including all following values for it. Executing the command resulted in the above mentioned error at line 690. It seems that you are comparing the number of values for each argument and in my case they will not match. I have found a workaround: I am supplying the one fan twice on the command line.

desbma, 2015-06-05:
Hi,

It is a bug, I was checking the wrong argument. 
It's fixed in version 1.1.3, you can update with `pip3 install -U hddfancontrol`.

Please tell me if everything now works fine!

reinersmann, 2015-06-05:
Everything seems to be running fine now - thanks!

closed 2015-06-05 by desbma