desbma/hddfancontrol #4: No module named 'daemon.pidlockfile'

vena, 2016-07-10:
This is odd, and I don't know enough python to know what's going wrong here... Running hddfancontrol manually from the command line works fine, but trying to start it at boot from rc.local, I get an exception.  This is on Ubuntu server 16.04, Python 3.5.1+

I'm running the same command from the command line and rc.local:
```
hddfancontrol \
        --drives /dev/sda /dev/sdb /dev/sdc \
        --pwm /sys/class/hwmon/hwmon1/device/pwm2 \
        --pwm-start-value 112 \
        --pwm-stop-value 64 \
        --min-fan-speed-prct 50 \
        --max-temp 45 \
        --hddtemp-daemon \
        --interval 30 \
        --background \
        --log-file /var/log/hddfancontrol.log
```

Here's the exception reported when run from rc.local:
```
Jul  9 22:06:32 forest rc.local[1217]: Traceback (most recent call last):
Jul  9 22:06:32 forest rc.local[1217]:   File "/usr/local/bin/hddfancontrol", line 7, in <module>
Jul  9 22:06:32 forest rc.local[1217]:     from hddfancontrol import cl_main
Jul  9 22:06:32 forest rc.local[1217]:   File "/usr/local/lib/python3.5/dist-packages/hddfancontrol/__init__.py", line 28, in <module>
Jul  9 22:06:32 forest rc.local[1217]:     import daemon.pidlockfile
Jul  9 22:06:32 forest rc.local[1217]: ImportError: No module named 'daemon.pidlockfile'
```

python3 and python3-daemon are installed.

Any ideas would be much appreciated!

desbma, 2016-07-10:
Hi,

How did you install hddfancontrol ?

The `python3-daemon` Ubuntu package is not needed, and is different (although they serve the same purpose) from the [PyPI package](https://pypi.python.org/pypi/python-daemon-3K/1.5.8) used by hddfancontrol.

If you follow the install instructions, PIP should install it automatically for you.

vena, 2016-07-10:
```sudo pip3 install hddfancontrol```

only reason i installed python3-daemon was in trying to hunt down the error when run from rc.local.  i assume there's some kind of environment issue, but hddfancontrol works from a root console just fine.

vena, 2016-07-10:
ok! figured one part out.  environment was the right idea, turns out python has no idea where to search for modules at the time rc.local is executed.  solution was to first get a working list of python's import paths...

```
import sys
sys.path
```

which gives me something like...

```'', '/usr/lib/python35.zip', '/usr/lib/python3.5', '/usr/lib/python3.5/plat-x86_64-linux-gnu', '/usr/lib/python3.5/lib-dynload', ... '/usr/local/lib/python3.5/dist-packages', '/usr/lib/python3/dist-packages']```

so my new rc.local entry looks like...

```
export PYTHONPATH="${PYTHONPATH}:/usr/lib/python3.5:/usr/lib/python3.5/plat-x86_64-linux:...
hddfancontrol ...
```

and now it can find deamon.pidlockfile.  except now it starts before hddtemp daemon and immediately exits since it can't connect.  oh well, i guess the answer to that is "use a real init script, not rc.local" :)

desbma, 2016-07-10:
> except now it starts before hddtemp daemon and immediately exits since it can't connect. oh well, i guess the answer to that is "use a real init script, not rc.local" :)

As a workaround, you could also remove the ` --hddtemp-daemon` flag from your command line. Hddfancontrol would spawn a "one shot" hddtemp process for each temperature reading instead of asking the daemon, but that should do the trick.

closed 2016-07-24 by desbma