desbma/hddfancontrol #6: Parameter fail

diluex, 2017-02-23:
ls@linkstation:~$ sudo hddfancontrol -d /dev/sda -p /sys/class/hwmon/hwmon0/pwm1                                                                                        2017-02-23 16:45:05,268 WARNING [Startup] Missing --pwm-start-value or --pwm-stop-value argument, running hardware test to find values
2017-02-23 16:45:05,353 INFO [TestHardware] Running hardware tests, this may take a few minutes
Test of drive sda WDC WD10EADS-00L5B1 ----------------------------------------------------------------------------------------------------------------------------------
Getting drive power state                                                                                                                                         [ OK ]
Getting drive temperature                                                                                                                                         [ KO ]
Getting drive activity statistics                                                                                                                                 [ OK ]
Test of fan #1 ---------------------------------------------------------------------------------------------------------------------------------------------------------
Stopping fan                                                                                                                                                      [ OK ]
Starting fan                                                                                                                                                      [ OK ]
Finding exact start value of fan                                                                                                                                  [ OK ]
Finding exact stop value of fan                                                                                                                                   [ OK ]
1/7 tests failed!
2017-02-23 16:45:47,630 INFO [TestHardware] Recommended parameters: --pwm-start-value 48 --pwm-stop-value 0
ls@linkstation:~$ sudo hddfancontrol -d /dev/sda -p /sys/class/hwmon/hwmon0/pwm1 --pwm-start-value 48 --pwm-stop-value 0
2017-02-23 16:46:19,748 INFO [Main] Process real time scheduler set to 2, priority 99
2017-02-23 16:46:19,866 ERROR [Main] ValueError: invalid literal for int() with base 10: ''

desbma, 2017-02-24:
Can you please post here the output of:
* `sudo hddtemp -u C -n /dev/sda`
* `nc 127.0.0.1 7634` (if you are running the hddtemp daemon)
* `sudo hdparm -H /dev/sda` (might report an error, it's OK)

desbma, 2017-02-24:
It should now be fixed, please [test by installing from sources](https://github.com/desbma/hddfancontrol#from-source) and confirm that the issue is gone.

diluex, 2017-02-25:
I've reinstalled it with source, but same error.

root@linkstation:~ /hddfancontrol# hddfancontrol -d /dev/sda -p /sys/class/hwmon/hwmon0/pwm1  --pwm-start-value 48 --pwm-stop-value 0
2017-02-25 12:10:21,978 INFO [Main] Process real time scheduler set to 2, priority 99
SG_IO: bad/missing sense data, sb[]:  70 00 05 00 00 00 00 0a 04 51 40 00 21 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
2017-02-25 12:10:22,143 ERROR [Main] ValueError: invalid literal for int() with base 10: ''
2017-02-25 12:10:22,147 INFO [Fan #1] Setting fan speed to 100%

desbma, 2017-02-25:
Sorry, should be fixed now.

Please test again from git master source.

diluex, 2017-02-25:
It's working smoothly now.
Great work and only solution for my Linkstation.
Thank your very much!

desbma, 2017-02-25:
Great

commit 6e13ecd6 (2017-02-24): Improve hdparm error handling (should fix #6)
closed 2017-02-25 by desbma