desbma/hddfancontrol #8: ERROR [Main] ValueError: invalid literal for int() with base 10: ''

nightah, 2018-02-02:
It appears that randomly I receive the aforementioned error and all of my controlled fans subsequently spin up to 100%, here's a log snippet:
```
2018-02-02 17:03:45,345 INFO [Main] Maximum drive temperature: 36 C
2018-02-02 17:04:01,786 INFO [Main] Maximum drive temperature: 36 C
2018-02-02 17:04:29,326 INFO [Main] Maximum drive temperature: 36 C
2018-02-02 17:04:49,757 INFO [Main] Maximum drive temperature: 36 C
2018-02-02 17:05:13,266 INFO [Main] Maximum drive temperature: 36 C
2018-02-02 17:05:29,478 INFO [Main] Maximum drive temperature: 36 C
2018-02-02 17:06:05,429 INFO [Main] Maximum drive temperature: 36 C
2018-02-02 17:06:44,600 ERROR [Main] ValueError: invalid literal for int() with base 10: ''
2018-02-02 17:06:44,601 INFO [Fan #1] Setting fan speed to 100%
2018-02-02 17:06:44,603 INFO [Fan #2] Setting fan speed to 100%
```

Everything will work normally if the service is restarted until again, randomly the error appears.

Running with the following options:
```
-d /dev/sda /dev/sdb /dev/sdc /dev/sdd /dev/sde /dev/sdf /dev/sdg /dev/sdh /dev/sdi /dev/sdj /dev/sdk /dev/sdl -p /sys/class/hwmon/hwmon3/pwm1 /sys/class/hwmon/hwmon3/pwm2 --pwm-start-value 128 32 --pwm-stop-value 80 0 --min-temp 30 --max-temp 42 --min-fan-speed-prct 10 -i 15 -b -l /var/log/hddfancontrol.log
```
HDD Fan Control v.1.2.7 on Arch Linux.

Example of logs after a restart:
```
2018-02-02 20:43:48,877 INFO [Main] Process real time scheduler set to 2, priority 99
2018-02-02 20:43:48,912 WARNING [sda ST4000DM000-1F2168] Drive does not allow querying temperature without going out of low power mode.
2018-02-02 20:43:48,918 WARNING [sdb ST4000DM000-1F2168] Drive does not allow querying temperature without going out of low power mode.
2018-02-02 20:43:48,927 WARNING [sdc ST4000DM000-1F2168] Drive does not allow querying temperature without going out of low power mode.
2018-02-02 20:43:48,936 WARNING [sdd ST4000DM000-1F2168] Drive does not allow querying temperature without going out of low power mode.
2018-02-02 20:43:48,956 WARNING [sde ST4000DM000-1F2168] Drive does not allow querying temperature without going out of low power mode.
2018-02-02 20:43:48,987 WARNING [sdf ST4000DM004-2CV104] Drive does not allow querying temperature without going out of low power mode.
2018-02-02 20:43:49,046 WARNING [sdg ST4000DM004-2CV104] Drive does not allow querying temperature without going out of low power mode.
2018-02-02 20:43:49,056 WARNING [sdh ST4000DM004-2CV104] Drive does not allow querying temperature without going out of low power mode.
2018-02-02 20:43:49,063 WARNING [sdi ST4000DM000-1F2168] Drive does not allow querying temperature without going out of low power mode.
2018-02-02 20:43:49,073 WARNING [sdj ST4000DM000-1F2168] Drive does not allow querying temperature without going out of low power mode.
2018-02-02 20:43:49,089 WARNING [sdk ST4000DM000-1F2168] Drive does not allow querying temperature without going out of low power mode.
2018-02-02 20:43:49,099 WARNING [sdl ST4000DM000-1F2168] Drive does not allow querying temperature without going out of low power mode.
2018-02-02 20:43:50,435 INFO [Main] Maximum drive temperature: 31 C
2018-02-02 20:43:50,435 INFO [Fan #1] Setting fan speed to 8%
2018-02-02 20:43:50,437 WARNING [Fan #1] /sys/class/hwmon/hwmon3/pwm1_enable was 0, setting it to 1
2018-02-02 20:43:50,438 INFO [Fan #2] Setting fan speed to 8%
2018-02-02 20:43:50,438 WARNING [Fan #2] /sys/class/hwmon/hwmon3/pwm2_enable was 0, setting it to 1
2018-02-02 20:44:06,856 INFO [Main] Maximum drive temperature: 31 C
2018-02-02 20:44:23,117 INFO [Main] Maximum drive temperature: 31 C
2018-02-02 20:44:39,519 INFO [Main] Maximum drive temperature: 31 C
```

desbma, 2018-02-03:
It seems the temperature query fails randomly for at least one of your drives, and hddtemp returns an invalid temperature.

To investigate further I would first check all your drives do not have any recorded SATA error: `sudo smartctl -H -l selftest -l error /dev/sdX`.

Then if you are using the hddtemp daemon (check with `systemctl status hddtemp`), run the following script in bash (you may need to install the Arch [`gnu-netcat`](https://www.archlinux.org/packages/extra/x86_64/gnu-netcat/) package):
`while true; do nc 127.0.0.1 7634 >> /tmp/hddtemp.log; sleep 10; done`

If you not running the daemon, run this instead:
`while true; do hddtemp /dev/sda >> /tmp/hddtemp.log; hddtemp /dev/sdb >> /tmp/hddtemp.log; hddtemp /dev/sdc >> /tmp/hddtemp.log; (repeat for all your drives) sleep 10; done`

After a while (depending on how long you reproduce the issue usually with hddfancontrol), you can post the content of /tmp/hddtemp.log here, to see if this is indeed an issue with the output of hddtemp.

nightah, 2018-02-03:
None of the drives have any reported errors so we can definitely rule that out.

I'm now running the script to output the temps to a file, so when the issue recurs I'll update again.

desbma, 2018-02-14:
Any chance at reproducing a failed temperature reading?

nightah, 2018-02-14:
Been monitoring for almost 2 weeks now, have not had the issue pop up again so I'll close this off.
Thanks again!

closed 2018-02-14 by nightah