desbma/hddfancontrol #9: ERROR [DriveSpinDownThread-sdb1 HUP722020APA330] FileNotFoundError: [Errno 2] No such file or directory: '/sys/block/sdb1/stat'

adrfantini, 2018-02-03:
Hi,
I have installed this neat little tool on Arch.

I have the error in the topic title:
`ERROR [DriveSpinDownThread-sdb1 HUP722020APA330] FileNotFoundError: [Errno 2] No such file or directory: '/sys/block/sdb1/stat'`

And the drives do not seem to spin down ever. This happens for different drives, a WDC green and 2 Hitachi drives. I did not test with other drives.

The files `'/sys/block/sdb1/stat'` exist. This is how I call hddfancontrol:

```
hddfancontrol -d /dev/disk/by-partuuid/4f7fba02-0410-444f-ae83-6bde9d15fffc /dev/disk/by-partuuid/5ff5b33a-b30f-4fb7-9703-9cbeaf9f466d /dev/disk/by-partuuid/caede4f9-f16e-481b-a979-ca3a63dcb59b --min-temp 35 -p /sys/devices/platform/it87.2624/hwmon/hwmon0/pwm2 --pwm-start-value 120 --pwm-stop-value 90 --min-fan-speed-prct 0 -i 60 --spin-down-time 600 -b -l /var/log/hddfancontrol.log --hddtemp-daemon
```

desbma, 2018-02-03:
Hi,

You are calling hddfancontrol with partition devices (`/dev/disk/by-partuuid/4f7fba02-0410-444f-ae83-6bde9d15fffc` which points to `/dev/sdb1`) instead of block devices (`/dev/sdb`).

You need to replace the `/dev/disk/by-partuuid/XXX` parts in your command line by something like `/dev/disk/by-id/XXX` (without any `-partX` suffix, because those actually links to the partition device and not the root block device) or `/dev/sdX`

desbma, 2018-02-03:
Also, even if this is unrelated to your issue, If you are using the Arch AUR package, you may be interested by the systemd service to automatically handle start/stop/logging of hddfancontrol.

You can put the parameters to pass to hddfancontrol in `/etc/conf.d/hddfancontrol` and then control the service with :
```
# start daemon
sudo systemctl start hddfancontrol
# auto start daemon at boot
sudo systemctl enable hddfancontrol
# see daemon status and last log lines
sudo systemctl status hddfancontrol
```

adrfantini, 2018-02-04:
You are of course absolutely right. Yes, I am using the systemd service, which is very handy.
Seems to work now, even though there is still some process waking up the disk from time to time... maybe smartd. But unrelated to hddfancontrol. Thanks!

adrfantini, 2018-02-04:
The drives do not throw out any error but they do not spin down after the specified time (right now 30s, for testing). How does hddfancontrol detect activity, so that I can do the same and see what it detects?

desbma, 2018-02-04:
It reads `/sys/block/sdX/stat`, if there is some activity, the numbers change.

closed 2018-02-04 by desbma

adrfantini, 2018-02-04:
Unfortunately I do not see any change, but the drives do not spin down. example for SDA, which is `/dev/disk/by-id/ata-HUP722020APA330_BFJ0WS3F`:

```
(fish shell) # while true
                      echo -ne (date); cat /sys/block/sda/stat ; sleep 30s
                  end
Sun Feb  4 13:43:37 CET 2018     189        0    12944    14046        0        0        0        0        0    13980    14046
Sun Feb  4 13:44:07 CET 2018     189        0    12944    14046        0        0        0        0        0    13980    14046
Sun Feb  4 13:44:37 CET 2018     189        0    12944    14046        0        0        0        0        0    13980    14046
Sun Feb  4 13:45:07 CET 2018     189        0    12944    14046        0        0        0        0        0    13980    14046
Sun Feb  4 13:45:37 CET 2018     189        0    12944    14046        0        0        0        0        0    13980    14046
Sun Feb  4 13:46:08 CET 2018     189        0    12944    14046        0        0        0        0        0    13980    14046
Sun Feb  4 13:46:38 CET 2018     189        0    12944    14046        0        0        0        0        0    13980    14046
Sun Feb  4 13:47:08 CET 2018     189        0    12944    14046        0        0        0        0        0    13980    14046
Sun Feb  4 13:47:38 CET 2018     189        0    12944    14046        0        0        0        0        0    13980    14046
Sun Feb  4 13:48:08 CET 2018     189        0    12944    14046        0        0        0        0        0    13980    14046
Sun Feb  4 13:48:38 CET 2018     189        0    12944    14046        0        0        0        0        0    13980    14046
Sun Feb  4 13:49:08 CET 2018     189        0    12944    14046        0        0        0        0        0    13980    14046
Sun Feb  4 13:49:38 CET 2018     189        0    12944    14046        0        0        0        0        0    13980    14046
Sun Feb  4 13:50:08 CET 2018     189        0    12944    14046        0        0        0        0        0    13980    14046
Sun Feb  4 13:50:38 CET 2018     189        0    12944    14046        0        0        0        0        0    13980    14046
Sun Feb  4 13:51:08 CET 2018     189        0    12944    14046        0        0        0        0        0    13980    14046
```

Hddfancontrol log file for the same period of time:
```
2018-02-04 13:41:46,528 INFO [Main] Maximum drive temperature: 42 C
2018-02-04 13:41:46,528 INFO [Fan #1] Setting fan speed to 46%
2018-02-04 13:42:46,559 INFO [Main] Maximum drive temperature: 42 C
2018-02-04 13:43:47,416 INFO [Main] Maximum drive temperature: 41 C
2018-02-04 13:43:47,416 INFO [Fan #1] Setting fan speed to 40%
2018-02-04 13:44:47,806 INFO [Main] Maximum drive temperature: 42 C
2018-02-04 13:44:47,806 INFO [Fan #1] Setting fan speed to 46%
2018-02-04 13:45:48,304 INFO [Main] Maximum drive temperature: 42 C
2018-02-04 13:46:48,695 INFO [Main] Maximum drive temperature: 42 C
2018-02-04 13:47:49,192 INFO [Main] Maximum drive temperature: 42 C
2018-02-04 13:48:49,592 INFO [Main] Maximum drive temperature: 41 C
2018-02-04 13:48:49,593 INFO [Fan #1] Setting fan speed to 40%
2018-02-04 13:49:50,092 INFO [Main] Maximum drive temperature: 42 C
2018-02-04 13:49:50,092 INFO [Fan #1] Setting fan speed to 46%
2018-02-04 13:50:50,851 INFO [Main] Maximum drive temperature: 42 C
2018-02-04 13:51:50,979 INFO [Main] Maximum drive temperature: 42 C
```
hddfancontrol command (verified via systemctl status):
```
HDDFANCONTROL_ARGS="-d /dev/disk/by-id/ata-HUP722020APA330_BFJ0WS3F /dev/disk/by-id/ata-HUP722020APA330_BFGWU7WF /dev/disk/by-id/ata-WDC_WD10EARS-00Y5B1_WD-WCAV5R693922 --min-temp 35 -p /sys/devices/platform/it87.2624/hwmon/hwmon0/pwm2 --pwm-start-value 120 --pwm-stop-value 90 --min-fan-speed-prct 0 -i 60 --spin-down-time 30 -b -l /var/log/hddfancontrol.log --hddtemp-daemon"
```

desbma, 2018-02-04:
You are using `-i 60 --spin-down-time 30` as parameters. The spin down time is lower that the check interval, in that case hddfancontrol does not do anything because the SATA configuration should be enough to spin down the drive.

However you hdparm command line does not set a 30s timeout, for that you would need `-S 6` (providing the drive respects the time, I know some drives don't.

> -S
>     Put the drive into idle (low-power) mode, and also set the standby (spindown) timeout for the drive. This timeout value is used by the drive to determine how long to wait (with no disk activity) before turning off the spindle motor to save power. Under such circumstances, the drive may take as long as 30 seconds to respond to a subsequent disk access, though most drives are much quicker. The encoding of the timeout value is somewhat peculiar. A value of zero means "timeouts are disabled": the device will not automatically enter standby mode. Values from 1 to 240 specify multiples of 5 seconds, yielding timeouts from 5 seconds to 20 minutes. Values from 241 to 251 specify from 1 to 11 units of 30 minutes, yielding timeouts from 30 minutes to 5.5 hours. A value of 252 signifies a timeout of 21 minutes. A value of 253 sets a vendor-defined timeout period between 8 and 12 hours, and the value 254 is reserved. 255 is interpreted as 21 minutes plus 15 seconds. Note that some older drives may have very different interpretations of these values. 

https://linux.die.net/man/8/hdparm

adrfantini, 2018-02-04:
Indeed that was the case. Thanks.
Maybe a note in the manual that the spin down time must be > than the interval to work would clarify further. In a case such as mine, where I just wanted to test, I failed to see why a spin down time of 90s worked but one of 30 did not.

Works absolutely wonderfully now.

desbma, 2018-02-04:
Yup, I'll add a note in the README about that, and also about the systemd service which is currently undocumented.