desbma/hddfancontrol #10: Logrotate and hddfancontrol

adrfantini, 2018-02-05:
I want to `logrotate` `hddfancontrol`'s output.
To do this I am writing a `logrotate` directive to go in `/etc/logrotate.d/hddfancontrol`.

Usually one kills the `logrotate`d process by `killall -HUP process` which, as far as I understand, instructs the process to reload the configuration and thus the log file.

I'm not sure if this works with `hddfancontrol`, as the daemon is killed (`ExecStart=/usr/bin/hddfancontrol $HDDFANCONTROL_ARGS (code=killed, signal=HUP)`) and I *think* it's still running. 

Is this the right approach for `logrotate`ing  `hddfancontrol`?

desbma, 2018-02-05:
If you use the systemd service you do not need logrotate at all, systemd handles both daemonization and logging, so you don't need `-b` or `-l` when using hddfancontrol with systemd.
You can see logs with `sudo journalctl -u hddfancontrol`.

If for some reason you don't want to use systemd's logging facility, and want to output to a text file, you can use the `-l` parameter, and don't need to send any signal when rotating the log file, because Python's [ WatchedFileHandler](https://docs.python.org/3/library/logging.handlers.html#watchedfilehandler) is used, so the file is automatically reopened if it is moved.

Hope that answers your questions.

adrfantini, 2018-02-06:
Ooh, that's nice! Thanks. Another one for the docs, I suppose. Good job!

closed 2018-02-06 by adrfantini

adrfantini, 2018-02-07:
Is it normal that all I get in the `journalctl` logs is lots of this:
```
-- Reboot --
Feb 03 21:02:23 serverino systemd[1]: Started HDD fan control.
Feb 03 21:14:00 serverino systemd[1]: Stopping HDD fan control...
Feb 03 21:14:00 serverino systemd[1]: Stopped HDD fan control.
```
?