desbma/hddfancontrol #12: Feature request: option to monitor CPU temperature too

adrfantini, 2018-04-06:
In small single fan systems it would be useful if HDDFC was able to monitor the CPU temperature too - otherwise the CPU temperature might get too high while the HDDs are turned off.

I can't think of any other solution for this, as using two fan control software at the same time seems like a very bad idea.

Would if be possible to add flags to HDDFC to set an additional sensor path to monitor, and its temperature targets?

desbma, 2018-04-07:
This is a reasonable request.

I personally use both hddfancontrol and fancontrol, because I have a separate fan close to the CPU on my NAS, so each program controls its own set of fans.
However I understand why you would need a single program to probe and maintain all temperatures  if you have a limited number of fans to cool all components.

Also the historic fancontrol program is actually a simple Bash script which does a sleep loop under normal priority. That might work well in most cases, but if your system is under 100% CPU usage or with almost no free memory, there is no guarantee that the 5s probe interval will not become 30s because of the system load, which can let your CPU temperature go out of control and can cause a brutal shutdown if it becomes too high.

desbma, 2019-01-26:
I have made a [cpu-temp](https://github.com/desbma/hddfancontrol/tree/cpu-temp) banch to test this experimental feature (currently testing on my own NAS).
The CPU sensor is passed with the `-c` parameter, and target temperature range is computed automatically for the CPU.

adrfantini, 2019-01-28:
Thank you! How do you compute ideal CPU temperatures? I would argue that people have different views on how hot their CPUs should be and that having target temperatures for the CPU too would be better. Some people for example would prefer to set a lower CPU temp threshold if they are also worried about VRM temps.
Now that I think of it, technically one could pass any sensor to the `-c` argument, not just the CPU. Why not make it more general? You would need to pass sensor path, min threshold, max threshold, and optionally additive and multiplicative scaling factors (which some sensors require).

desbma, 2019-01-28:
> Thank you! How do you compute ideal CPU temperatures? I would argue that people have different views on how hot their CPUs should be and that having target temperatures for the CPU too would be [...]

Most (all ?) sensors in sysfs report a `max` and `crit` temperature. Max is the maximum long term operating temperature, and crit is what you should never ever reach because that can cause permanent component damage (most CPU automatically shut down at this temperature for security).

What I do is get max, and remove a margin for safety: https://github.com/desbma/hddfancontrol/blob/9006a1a44d91f1b1a98b92b4ab1e90508ab4021e/hddfancontrol/__init__.py#L306-L325
I used that mostly because I'm lazy and this is something I quickly hacked together to test it. But yes the idea in the end is to pass a min and max target temp, like for hard drive.

> Now that I think of it, technically one could pass any sensor to the `-c` argument, not just the CPU. Why not make it more general? You would need to pass sensor path, min threshold, max threshold, and optionally additive and multiplicative scaling factors (which some sensors require).

You can already pass any motherboard sensor here. I should probably rename the parameter to make it more generic. Beware though some sensors are buggy and report incorrect temperatures.

adrfantini, 2019-01-28:
Excellent. I can test this if you need it, I guess I could just curl the init file?

desbma, 2019-01-28:
Curl the `__init__.py`
or 
[install from source](https://github.com/desbma/hddfancontrol#from-source) (change branch with `git checkout cpu-temp`), and in that case remove the AUR package previously to avoid conflicts

adrfantini, 2019-01-29:
Tested and works for me, at least using coretemp. To use this, I would personally prefer to be able to set temperature thresholds myself, since I do not want to start the fan before reaching, say, 50C.

desbma, 2019-01-30:
If you want to quickly test with 50 °C instead of 30, this is literally a single character change: https://github.com/desbma/hddfancontrol/blob/c4c8aaf6971b4a6f7f6de9bfc6f3c8b07788c1d3/hddfancontrol/__init__.py#L281.

adrfantini, 2019-01-30:
Yup, testing with higher temps right now.

desbma, 2019-09-21:
I am closing this, since I have added a `--cpu-temp-range` argument, and merged all in the `master` branch, so I consider the feature complete.

closed 2019-09-21 by desbma

adrfantini, 2019-09-23:
I'll try to test it asap, thank you very much!

wiesion, 2019-09-28:
This is a good idea and a welcome feature.

But consider the scenario of a separated HDD bay (For cube cases, e.g. Fractal Design Node 804) or when SATA enclosures are applied. In such cases you need the ability to be able to control different fans in different areas for either CPUs or HDDs. 

For instance, the fans connected to pwm1 through pwm3 should cool the HDDs based on their temperatures, while the fans connected to pwm4 through pwm6 should cool the CPU given a probe filepath.

PS: Even more advanced use-cases would be multiple HDD regions and multiple CPU cooling

desbma, 2019-09-28:
> But consider the scenario of a separated HDD bay (For cube cases, e.g. Fractal Design Node 804) or when SATA enclosures are applied. In such cases you need the ability to be able to control different fans in different areas for either CPUs or HDDs.

This can already be done using both `hddfancontrol` and [`fancontrol`](https://linux.die.net/man/8/fancontrol). I actually have been using this for several years without issues, as long as you have separate fans, that works well.
The reason I have added this feature is to support the opposite use case : different heat sources (CPU and HDD), but a single fan.

> PS: Even more advanced use-cases would be multiple HDD regions and multiple CPU cooling

That can also be done currently by starting several `hddfancontrol` instances, as long as the controlled fans do not overlap. You probably need to tweak the systemd unit files to be able to start more than one instance. I have not tested this, but I see no reason why that would not work.

SHU-red, 2022-07-09:
Sorry for this question here but i am using hddfancontrol via docker and it works like a charm.
Since ive recently been using my homeserver also to transcode video, i ran into problems with high CPU temperatures.
Could someone give me a hint what to do with my docker-compose-file to also monitor CPU?
I dont care if this mix of CPU and HDD temperatures may also lead to permanent higher fan speeds.
Im just not very into this topic and am wondering if there is a way to just add the device CPU similar to my hdds:

```
version: "3"
services:
  hddfancontrol:
    image: fred92/hddfancontrol:master
    container_name: hddfancontrol
    restart: unless-stopped
    volumes:
      - /lib/modules:/lib/modules:ro
    privileged: true
    cap_add:
      - SYS_MODULE
    environment:
      - DEVICES=/dev/sda1 /dev/sdb1 /dev/sdc1 /dev/sdc2 #POSSIBLE TO ADD CPU HERE???
      - PWM_DEVICES=/sys/class/hwmon/hwmon2/pwm1 /sys/class/hwmon/hwmon2/pwm2
      - PWM_START=70 80
      - PWM_STOP=20 30
      - MIN_TEMP=30
      - MAX_TEMP=50
      - MIN_FAN=15
      - INTERVALL=60
      - SPINDOWN_TIME=900
      - LOG_PATH=/var/log/hddfancontrol.log
```