desbma/hddfancontrol #16: ERROR [Main] ValueError: max() arg is an empty sequence

adrfantini, 2019-01-30:
I got this error today; it was the first time I saw it and cannot reproduce it, as far as I can tell. I am using the cpu-temp branch, but no `-c` option right now.
Here are some additional info:
```
HDDFANCONTROL_ARGS="-d /dev/disk/by-id/ata-ST3000DM001-9YN166_S1F0B4GH /dev/disk/by-id/ata-ST3000DM001-9YN166_W1F0RPN8 /dev/disk/by-id/ata-WDC_WD10EARS-00Y5B1_WD-WCAV5R693922 --min-temp 35 --max-temp 40 -p /run/hddfancontrol_fan_hwmon/pwm2 --pwm-start-value 120 --pwm-stop-value 90 --min-fan-speed-prct 0 -i 60 --spin-down-time 600 --hddtemp-daemon"
```
```
nc localhost 7634 |sed 's/|//m' | sed 's/||/ \n/g' | awk -F'|' '{print $1 " " $3 " " $4}'
/dev/disk/by-id/ata-ST3000DM001-9YN166_S1F0B4GH 24 C 
/dev/disk/by-id/ata-ST3000DM001-9YN166_W1F0RPN8 23 C 
/dev/disk/by-id/ata-WDC_WD10EARS-00Y5B1_WD-WCAV5R693922 SLP *
```
```
hdparm -CH /dev/disk/by-id/{ata-ST3000DM001-9YN166_S1F0B4GH,ata-ST3000DM001-9YN166_W1F0RPN8,ata-WDC_WD10EARS-00Y5B1_WD-WCAV5R693922}

/dev/disk/by-id/ata-ST3000DM001-9YN166_S1F0B4GH:
SG_IO: bad/missing sense data, sb[]:  70 00 05 00 00 00 00 0a 04 51 40 00 21 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 drive temperature (celsius) is:  -18
 drive temperature in range:  yes
 drive state is:  active/idle

/dev/disk/by-id/ata-ST3000DM001-9YN166_W1F0RPN8:
SG_IO: bad/missing sense data, sb[]:  70 00 05 00 00 00 00 0a 04 51 40 00 21 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 drive temperature (celsius) is:  -18
 drive temperature in range:  yes
 drive state is:  active/idle

/dev/disk/by-id/ata-WDC_WD10EARS-00Y5B1_WD-WCAV5R693922:
SG_IO: bad/missing sense data, sb[]:  70 00 05 00 00 00 00 0a 04 51 40 00 21 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 drive temperature (celsius) is:  -18
 drive temperature in range:  yes
 drive state is:  standby
```
Logs:
```
...
gen 30 08:52:41 serverino hddfancontrol[454]: INFO [Main] Maximum device temperature: 23 °C
gen 30 08:53:41 serverino hddfancontrol[454]: INFO [Main] Maximum device temperature: 24 °C
gen 30 08:54:00 serverino hddfancontrol[454]: INFO [sda WDC WD10EARS-00Y5B1] Spinning down drive sda WDC WD10EARS-00Y5B1
gen 30 08:54:00 serverino hddfancontrol[454]: INFO [sdd ST3000DM001-9YN166] Spinning down drive sdd ST3000DM001-9YN166
gen 30 08:54:00 serverino hddfancontrol[454]: INFO [sdb ST3000DM001-9YN166] Spinning down drive sdb ST3000DM001-9YN166
gen 30 08:54:41 serverino hddfancontrol[454]: INFO [Main] All drives are in low power state
gen 30 08:54:41 serverino hddfancontrol[454]: ERROR [Main] ValueError: max() arg is an empty sequence
gen 30 08:54:41 serverino hddfancontrol[454]: INFO [Fan #1] Setting fan speed to 100%
gen 30 08:54:41 serverino hddfancontrol[454]: INFO [DriveSpinDownThread-sda WDC WD10EARS-00Y5B1] Exiting
gen 30 08:54:41 serverino hddfancontrol[454]: INFO [DriveSpinDownThread-sdd ST3000DM001-9YN166] Exiting
gen 30 08:54:41 serverino hddfancontrol[454]: INFO [DriveSpinDownThread-sdb ST3000DM001-9YN166] Exiting
```

desbma, 2019-01-30:
Thanks for the report, it should be fixed in https://github.com/desbma/hddfancontrol/commit/c4c8aaf6971b4a6f7f6de9bfc6f3c8b07788c1d3.

closed 2019-01-30 by desbma

adrfantini, 2019-01-30:
Seems to work now!

commit c4c8aaf6 (2019-09-21): Fix #16