desbma/hddfancontrol #17: pwm_enable resets to 2 after suspend

adrfantini, 2019-02-07:
When `hddfancontrol` starts, I get this warning:
```
WARNING [Fan #1] /run/hddfancontrol_fan_hwmon/pwm2_enable was 2, setting it to 1
```
After a suspend-resume, the kernel apparently resets the value to 2, and `hddfancontrol` is not able to control fan speed anymore. In fact, if I manually try:
```
root@serverino ~# echo 120 >  /run/hddfancontrol_fan_hwmon/pwm2
Error while writing to stdout
write_loop: Device or resource busy
root@serverino ~# echo 1 >  /run/hddfancontrol_fan_hwmon/pwm2_enable 
root@serverino ~# echo 120 >  /run/hddfancontrol_fan_hwmon/pwm2
root@serverino ~# 
```

Maybe `hddfancontrol` should periodically check the `pwm_enable` file, instead of just at startup?

desbma, 2019-02-07:
The value of `pwmX_enable` is checked, and set to 1 if needed, everytime a value is written to the PWM file.
What error are you having after resuming?

adrfantini, 2019-02-07:
I got no errors in the logs and stuff seems to function normally. I guess that when I was looking, `hddfancontrol` did not need to turn the fan on, so I did not notice it resetting the value. I assumed, wrongly, that it did so only after startup. Sorry for wasting your time, closing.

closed 2019-02-07 by adrfantini