desbma/hddfancontrol #18: IPMI Support

nightah, 2019-02-16:
Considering that several server motherboards only allow fan control via IPMI is there any chance we can get hddfancontrol to interface either via `ipmitool` or alternative tool so that it can adjust fan speeds?

Unfortunately per the [hwmon wiki](https://hwmon.wiki.kernel.org/device_support_status) while sensors-detect can recognise the driver there is no integration, there are no controllable pwm sensors exposed via lm-sensors.

I've tried doing some research but it doesn't look like there are any alternatives that would quite fill the gap that hddfancontrol covered.

Happy to provide any additional details, support or testing you might need.

desbma, 2019-02-16:
I don't know much about IMPI, but would be open to add support for fans or sensors though IPMI.

However I don't have any hardware with IPMI right now. So someone would have to step in to write the code to add support for it, test it (ideally on different pieces hardware), and do minimal maintenance for that part (mostly ensuring that nothing breaks with new Linux/ ipmitool / hddfancontrol releases).

nightah, 2019-02-19:
Thanks for the response @desbma, it appears that my board's IPMI actually doesn't allow control of the fans outside of the BIOS (thanks Gigabyte). While this isn't the norm, it means even if we solved this problem it wouldn't work for me.

Prior to me closing this issue off I think it's worthwhile asking the broader question of whether interfacing with other tools to control fans is something you might be interested in achieving for this project.

For example the solution I'm looking at now is buying a Corsair Commander Pro to give me a fan controller that I can then control with it's supplied software, in this case because of Linux, the open source varaint [OpenCorsairLink](https://github.com/audiohacked/OpenCorsairLink).

The relevant options to control fans are the following:
```
        --device <Device Number>        :Select device.
        
        Fan:
        --fan channel=N,mode=N,pwm=PWM,rpm=RPM,temps=TEMP:TEMP:TEMP,speeds=SPEED:SPEED:SPEED
                Channel: <fan number> :Selects a fan to setup. Accepted values are 1, 2, 3 or 4.
                Modes:
                         0 - Fixed PWM (requires to specify the PWM)
                         1 - Fixed RPM (requires to specify the RPM)
                         2 - Default
                         3 - Quiet
                         4 - Balanced
                         5 - Performance
                         6 - Custom Curve
                PWM <PWM Percent>       :The desired PWM for the selected fan. NOTE: it only works when fan mode is set to Fixed PWM
                RPM <fan RPM>   :The desired RPM for the selected fan. NOTE: it works only when fan mode is set to Fixed RPM
                For Custom Curves:
                        Temps <C>       :Define Celsius Temperatures for Fan.
                        Speeds <Percentage>     :Define Values of RPM Percentage for Fan.
```

Could we have an option to specify an external tool (in this case `OpenCorsairLink`) and relevant device and fan number(s) for hddfancontrol to achieve it's goal? I would imagine the aforementioned details plus a PWM percent that hddfancontrol calculates wouldn't be too difficult?

Happy to lodge this as another issue, please let me know how you want me to proceed.

desbma, 2019-02-21:
Yes I suppose I could support controlling the fans though a custom command.

Passing `-p command:/usr/local/bin/custom_script` to hddfancontrol would instruct it to call `/usr/local/bin/custom_script` to control fan speed, using fan speed percentage as the first argument.

In the script you can then convert fan percentage to whatever value you need, and call any third party tool.

You can open a new issue, or rename this one.

nightah, 2019-02-21:
Done with #19.

closed 2019-02-21 by nightah