desbma/hddfancontrol #19: Fan control through third party scripts

nightah, 2019-02-21:
Further to #18 this ticket is to track support for interfacing with third party tools/scripts to control fans speeds.

nightah, 2019-03-19:
@desbma have you had a chance to dig in to this yet? :)

desbma, 2019-03-19:
Not yet, sorry.

LightTemplar, 2024-05-02:
Coming from IPMI issue, I find too calling custom scripts as good universal solution. Moreover, this could be also a solution for a case, where HDDs are been controlled by TrueNAS via HBA card, and TrueNAS itself is placed as Virtual Machine inside Proxmox.

desbma, 2026-02-12:
7 years later, here we go: 307d2d1dc2de1321587f55cfd81c12e40eeda864

```console
$ hddfancontrol daemon -h
Start fan control daemon

Usage: hddfancontrol daemon [OPTIONS] --drives <DRIVES>... <--pwm <PWM>...|--fan-cmd <FAN_CMD>...>

Options:
[...]
  -f, --fan-cmd <FAN_CMD>...
          Control fan speed through external command. This command will be called to set or change fan speed with the target speed passed as the first argument, as an integer between 0 and 1000, where 0 should stop the fan, and 1000 set it to its maximum speed
```

Please test!

closed 2026-02-28 by desbma