desbma/hddfancontrol #22: Improvement: Average temperature over the last X minutes

tobiaspc, 2019-06-28:
Hi,

thanks for this great piece of software! I have one suggestion though:

My server has multiple WD Reds cooled by two fans. Both fans are connected to the same fan header on the motherboard. My configuration looks like this:
Min speed: 30%
Min temp: 30 deg
Max temp: 42 deg

In general, all my drives are between 36 and 40 deg hot. Since this range is so tight, when one drives temperature rises one degree, the fan speed jumps about 100rpm higher. After a few minutes, the temps drop by one deg and the fans spin down 100rpm. This results in constantly jumping (audible jumping, as the delta is so large) fan speeds: See the attached image from grafana.

[image]

Suggestion: Store the temperatures from the last X (configurable) readings and calculate an average temperature (as float). Then adjust the fan speed to the calculated average. 

The root cause is of cause the fact that the drives report their temps rounded to full degrees. My suggestion would serve as a workaround.

desbma, 2019-06-28:
I think you are trying to solve a problem that does not exist.

If you configure a very narrow temperature range, yes the fans will have perceptible jumps in RPM, but how is that a problem exactly?

And why are you setting such a low maximum temperature when most drives have a much higher operating range?

tobiaspc, 2019-06-28:
Thanks for the fast reply. Keeping the drives at or below 40 deg is just a personal preference. You would suggest a wider range and therefore hotter temperatures and lower fan jumps?

desbma, 2019-06-28:
What you are describing is not a problem, it is just the only logical consequence of the very narrow temperature range you have configured.

If you really want a narrow temperature range, then there will be fan speeds jumps, so what? The temperature should be maintained in the range you have configured as expected.

If you want smoother fan speed transitions, then just configure a larger temperature range, and speed jumps will disappear.

Everyone can have specific needs or preference, and I am just providing a tool that is configurable to support them.

Personally I configure the maximum drive temperature temperature minus 5-10 degrees to have a security margin, and minimize noise, but that is only my preference.

tobiaspc, 2019-06-29:
Okay many thanks for the hints!

closed 2019-06-29 by tobiaspc