desbma/hddfancontrol #23: TypeError thrown in 1.3.0

c4tz, 2019-11-09:
Hey,

first of all: thank you for this neat piece of software! It is exactly what I needed for my NAS and I've been happily using it for the past half year.

Now, I've just upgraded to `1.3.0` (using the AUR package) and get the following error:

```python
Traceback (most recent call last):
    File "/usr/bin/hddfancontrol", line 11, in <module>
        load_entry_point('hddfancontrol==1.3.0', 'console_scripts', 'hddfancontrol')()
    File "/usr/lib/python3.7/site-packages/hddfancontrol/__init__.py", line 976, in cl_main
    if (args.spin_down_time_s is not None) and (args.spin_down_time_s < args.interval_s):
TypeError: '<' not supported between instances of 'int' and 'NoneType'
```

I have the following params in my `/etc/conf.d/hddfancontrol`:

```
HDDFANCONTROL_ARGS="-d /dev/sdb /dev/sdc /dev/sdd -p /sys/class/hwmon/hwmon2/pwm1 /sys/class/hwmon/hwmon2/pwm2 --pwm-start-value 60 60 --pwm-stop-value 10 10 --min-fan-speed-prct 0 --spin-down-time 600"
```

I rolled back to 1.2.10 and everthing seems to be fine and working again. I didn't have time to look at the changes, but something must have been broken since then.

desbma, 2019-11-09:
Thanks for the report, it's fixed now.

Until the next release, you can work around this by adding `-i 60` as command line arguments.

closed 2019-11-09 by desbma

c4tz, 2019-11-10:
Wow, that literally was a "quick fix". Thank you!

commit 0811570c (2019-11-09): Argument handling fix (fixes #23)