desbma/hddfancontrol #24: No HDD spindown

c4tz, 2020-02-24:
Hey, its me again.

I updated my (NAS-)System yesterday. After that I rebooted and noticed that one of my drives won't spin down anymore while I'm running hddfancontrol, keeping my fans (and its LEDs) on, enlightening my bedroom :/

If I spin the HDD down manually (with `hdparm -Y`), it will be woken up by hddfancontrol after about 10 seconds. If I spin it down without hddfancontrol running, it will stay in standby mode. Only if I put it to sleep with `hdparm -y` and let hddfancontrol run, it won't be woken up again.

The real issue is the not spinning down, though, as hddfancontrol uses `hdparm -y`. It seems to me that the `hdparm -C` call keeps it running. I also already  tried `--smartctl`, but had no luck with that.

Do you have a suggestion on what could be the problem?

Version: `1.3.1`
OS: Arch Linux
Params: `-d /dev/sda /dev/sdc -p /sys/class/hwmon/hwmon2/pwm1 /sys/class/hwmon/hwmon2/pwm2 --pwm-start-value 60 60 --pwm-stop-value 10 10 --min-fan-speed-prct 0 --spin-down-time 60`

desbma, 2020-02-24:
Since the issue is occurring after a software update, it is likely some program is generating drive activity which prevents the drive from spinning down.

If you stop hddfancontrol, then run:
```
cat /sys/block/sda/stat
cat /sys/block/sdc/stat
```
then wait 10 minutes and re-read the stats, are the numbers **exactly** the same?

c4tz, 2020-02-24:
Thank you for the quick response! I already figured out it was only `/dev/sda`, so I did it only for that drive.

```bash
$ cat /sys/block/sda/stat
   17043     1438  3226752   132658    14542     1251 17398256   313006        0    96214   410170        0        0        0        0       57     1400
$ cat /sys/block/sda/stat
   17043     1438  3226752   132658    14542     1251 17398256   313006        0    96214   410170        0        0        0        0       57     1400
```

I also tried `iotop`, `dstat` and `iosnoop` before, but they didn't show anything. Does a call to `hdparm -C` alter these values?

desbma, 2020-02-24:
> Does a call to hdparm -C alter these values?

It shouldn't but you can check with:
```
cat /sys/block/sda/stat
hdparm -C /dev/sda
cat /sys/block/sda/stat
```

c4tz, 2020-02-24:
Oh yeah, sorry. Could've come up with that myself.

```bash
$ cat /sys/block/sda/stat && sudo hdparm -C /dev/sda && cat /sys/block/sda/stat
   21935     1438  4478728   136708    14542     1251 17398256   313006        0   114297   412417
        0        0        0        0       57     1400

/dev/sda:
 drive state is:  active/idle
   21936     1438  4478728   136708    14542     1251 17398256   313006        0   114300   412417
        0        0        0        0       57     1400
```

It seems to alter them a little though. Could this be the reason it resets the idle counter?

desbma, 2020-03-01:
According to https://www.kernel.org/doc/Documentation/block/stat.txt the counters seem to indicate that a read request had been done.

I have no idea why `hdparm -C` would trigger a read, and why that behavior would have changed.

c4tz, 2020-03-01:
What kernel version do you use? I noticed the problem after upgrading from 5.3.x to 5.5.x. I could imagine some changes in the IO layer may be the cause.

desbma, 2020-03-01:
I'm on kernel 5.4.23.
The counters do not change when I run `hdparm -C`.

I'll test on 5.5.7.

c4tz, 2020-03-01:
Thanks! I could also post a list of all upgraded packages, if it doesn't change for you with 5.5.7

desbma, 2020-03-01:
My PC does not boot with kernel 5.5.7 on Arch Linux, so I can't check.

Can your revert to kernel 5.3.x to see if that changes something?

c4tz, 2020-03-01:
I sure can try to, but it might take some time (till Friday) as I'll be quite busy the next few days.

Thank you for the time you've put into this already!

c4tz, 2020-03-06:
Okay, so I downgraded my NAS to the package-state from 2019-11-09.

This is the result:

```bash
$ cat /sys/block/sda/stat && sudo hdparm -C /dev/sda && cat /sys/block/sda/stat
     460     1265    21728     2315       14        7      152       33        0      310     1980        0        0        0        0

/dev/sda:
 drive state is:  active/idle
     460     1265    21728     2315       14        7      152       33        0      310     1980        0        0        0        0
```

Afterwards, I only upgraded the kernel back to 5.5:

```bash
$ cat /sys/block/sda/stat && sudo hdparm -C /dev/sda && cat /sys/block/sda/stat
     416     1312    21730     1606        6        2       56        9        0      317     1267        0        0        0        0        3        7

/dev/sda:
 drive state is:  active/idle
     417     1312    21730     1606        6        2       56        9        0      320     1267        0        0        0        0        3        7
```

So it really seems to be a change in the Kernel/IO layer. How do we handle this?

desbma, 2020-03-06:
Well, if you have time to spare and are motivated to fix this you can:
* Try to track down what kernel change have caused this, and in which part (is this a driver change or something that affects all drives in the block subsystem?)
* Contact the maintainer of that subsystem, or create a bug report in the right place, and get it noticed (this is the hardest part in my experience)
* Hope this is actually considered a bug, and fixed. Although several programs use the same method I use to identify idle drives, and it seems common sense, there is no guarantee whatsoever that probing the drive state with the ioctl hdparm uses does not generate drive IO
* Wait a few months/years so that the fix lands is a stable kernel release that you can install on your NAS

Or a more pragmatic/lazy/coward choice: revert your kernel version to the previous one, and hope it gets noticed and fixed by someone else.

c4tz, 2020-03-07:
Thanks for the input. I chose the latter option, because I don't have that much time to spare, unfortunately. I switched to the LTS kernel and am now running on `5.4.24`.

I'll close this for now, if the problem persists till December 2021, I'll open it again ;)

Thank you again for your time and this great piece of software.

closed 2020-03-07 by c4tz