desbma/hddfancontrol #25: log entry every 20 seconds

amo13, 2020-04-19:
thank you a lot for this application!

I use it on archlinux with systemd. But even though I set --interval 3600, the logs are showing new lines every 20 seconds, appearing to poll the temperature every 20 seconds:

```
Apr 19 17:35:33 server hddfancontrol[148710]: INFO [Main] Maximum device temperature: 36 °C
Apr 19 17:35:53 server hddfancontrol[148710]: INFO [Main] Maximum device temperature: 36 °C
Apr 19 17:36:13 server hddfancontrol[148710]: INFO [Main] Maximum device temperature: 36 °C
Apr 19 17:36:33 server hddfancontrol[148710]: INFO [Main] Maximum device temperature: 36 °C
```

Does this actually mean that it is polling the smart values every 20 seconds? And if so, is there a way to prevent the logs from being filled up unnecessarily?

desbma, 2020-04-19:
There is a [special logic](https://github.com/desbma/hddfancontrol/blob/04471fc0b2ebdffd6da4ceabce67492c0f22c23b/hddfancontrol/__init__.py#L833-L836) for the first 5min after the drives have been spun up or started, because they can heat up quickly during that time. After that time, the interval should be respected.

If you are concerned about the amount of logs produces, you can also pass the `-v warning` arguments.

amo13, 2020-04-19:
Ah ok, now I see!
I appreciate your fast and precise answer =)

closed 2020-04-19 by amo13