desbma/hddfancontrol #26: Plans on docker image?

fightforlife, 2020-05-18:
Hi there,
I am using this tool for quite sometime now. And it was just working perfectly after some needed configurations. (E.g. that the service starts after hddtemp,..)
The latest update to Ubuntu 20.04 broke hddfancontrol. I am quite sure there is an easy fix. (Something regarding the scheduler)
But wouldn't it be even better put the configuration, dependencies and binaries in a docker container?

Are there any plans for a docker container?

Best regards!

desbma, 2020-05-18:
I don't intend to provide a docker container.

However if you can provide more information (at the very least the log output with `-v debug`) about what is wrong on Ubuntu 20.04, maybe I can fix it.

fightforlife, 2020-05-19:
```
● hddfancontrol.service - HDD fan control
     Loaded: loaded (/etc/systemd/system/hddfancontrol.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Tue 2020-05-19 20:55:42 CEST; 1min 7s ago
    Process: 11420 ExecStartPre=/bin/sleep 60 (code=exited, status=214/SETSCHEDULER)

May 19 20:55:42 server systemd[1]: Starting HDD fan control...
May 19 20:55:42 server systemd[11420]: hddfancontrol.service: Failed to set up CPU scheduling: Operation not permitted
May 19 20:55:42 server systemd[11420]: hddfancontrol.service: Failed at step SETSCHEDULER spawning /bin/sleep: Operation not permitted
May 19 20:55:42 server systemd[1]: hddfancontrol.service: Control process exited, code=exited, status=214/SETSCHEDULER
May 19 20:55:42 server systemd[1]: hddfancontrol.service: Failed with result 'exit-code'.
May 19 20:55:42 server systemd[1]: Failed to start HDD fan control.

```

fightforlife, 2020-05-19:
Got it working again. I am not really sure why I needed to set the scheduler here. I just commented it out. And it works again.

ExecStartPre=/bin/sleep 60

This line is needed because hddfancontrol starts before hddtemp? is there a better way to do this?

```
[Unit]
Description=HDD fan control

[Service]
EnvironmentFile=/etc/conf.d/hddfancontrol
ExecStartPre=/bin/sleep 60
ExecStart=/usr/local/bin/hddfancontrol $HDDFANCONTROL_ARGS

# Scheduler priority
#CPUSchedulingPolicy=rr
#CPUSchedulingPriority=49

# Hardening
ProtectSystem=strict
PrivateTmp=true
ProtectHome=true
SystemCallArchitectures=native
MemoryDenyWriteExecute=true
CapabilityBoundingSet=CAP_SYS_RAWIO
AmbientCapabilities=CAP_SYS_RAWIO
NoNewPrivileges=true

[Install]
WantedBy=multi-user.target

```

desbma, 2020-05-19:
> `Failed to set up CPU scheduling: Operation not permitted`

You are running this error, when you are running hddfancontrol in a container, right? This should not be possible otherwise.

> ExecStartPre=/bin/sleep 60

What error do you get exactly without this?

fightforlife, 2020-05-21:
I am currently not running hddfancontrol in a container. I get this error when running the service as root on the host. I am not sure why. I can reporduce it by commenting&uncommenting the scheduler part in the service definition.

I needed the sleep command because otherwise the hddfancontrol service is starting before hddtemp. And that causes hddfancontrol to not start porperly. It will go into an error state.

desbma, 2020-05-21:
I have added https://github.com/desbma/hddfancontrol/commit/02e83f7455c63e68bb7213df4169693d01e8a637 which should fix the service ordering.

For the `Operation not permitted` error, are you running a special hardened kernel, or with a security patch?

If you run `chrt -r 49 sleep 1` as roo, do you get an error?

fightforlife, 2020-05-22:
Hi, yes on a fresh install of ubuntu server 20.04 I get the following error:
```
root@server:~# chrt -r 49 sleep 1
chrt: failed to set pid 0's policy: Operation not permitted
```

I also created a very simple Dockerfile which is doing the job for me:
```
FROM python:slim-buster
RUN apt-get update
RUN apt-get install -y smartmontools hdparm hddtemp fancontrol
RUN pip3 install hddfancontrol
CMD hddfancontrol $ARGS
```
I am thinking about adding some things and then publish it in docker hub.
(hddtemp deamon, build hddfancontrol from source, ....)

desbma, 2020-05-23:
> Hi, yes on a fresh install of ubuntu server 20.04 I get the following error:

I just checked on a freshly installed Ubuntu 20.04 VM, and could not reproduce.
Are you sure you don't have any sort of non standard hardening or configuration?
Root user should be able to change scheduler.

The idea behind setting the scheduler to real time, is that if your system is under very high load, or if the memory of Hddfancontrol gets swapped to disk, the system will prioritize its execution before other processes, and your drives will remain properly cooled.

adrfantini, 2020-06-19:
Having a docker container would hugely help in getting hddfancontrol working on, for example, unraid systems. It'd be very cool indeed!

desbma, 2020-06-19:
As I said before, I don't intend to provide a docker container.

I only kept this issue open because of the permission error, but since it seems to be specific to @fightforlife's setup, I am closing this.

closed 2020-06-19 by desbma

adrfantini, 2020-06-20:
> Hi, yes on a fresh install of ubuntu server 20.04 I get the following error:
>
> ```
> root@server:~# chrt -r 49 sleep 1
> chrt: failed to set pid 0's policy: Operation not permitted
> ```
>
> I also created a [...]

@fightforlife 
I can help with testing the dockerfile if necessary!

fightforlife, 2020-06-29:
@adrfantini 
I created a first version of the docker image here. It is still kind of rudimentary, but working fine for me.
You need to configure lm-sensors on your host system before using this image. 
The hddfancontrol paramters can be provided by using the $ARGS environment variable

https://github.com/fightforlife/docker_hddfancontrol
https://hub.docker.com/r/fred92/hddfancontrol

adrfantini, 2020-06-30:
> @adrfantini
> I created a first version of the docker image here. It is still kind of rudimentary, but working fine for me.
> You need to configure lm-sensors on your host system before using this [...]

Thank you very much @fightforlife !