desbma/hddfancontrol #29: Weird fan speed bahaviour

mindrunner, 2020-12-31:
I am observing a weird behavior which I cannot really explain to myself. Sometimes the Fans seem to not start, even though hddfancontrol raises the PWM values. This seems to recover by itself after the next hdd-spindown cycle.

[image]
The log shows that hddfancontrol adjusts, but temps are constantly going up. 

At  the same time:
[image]

Abount 10:33pm, drives are going to standby (see Monitoring screenshot)

Shortly after, drives are spinning up again and, voila, fan control is working....
[image]
(see Monitoring screenshot)

For reference, the whole Monitoring:
[image]
(Sidenote: If PWM reports 366RPM, the fans are off, that seems to be an issue with the mainboard)

Later that night, it happened again:
[image]

On a longer timeline, we can this:

[image]

If fans are going up normally, the disk temps are about 40C, but in some occasions when the fans are not coming up, the temps are going to >55C :(

I recently swapped the Mainboard from an ASUS P8P67 LE to a MSI Z77 MPOWER. I have had this issue on both mainboards.

I am not sure if this is a hddfancontrol bug or something else, but I am a little bit stuck and don't know what to do :/

Any ideas?

desbma, 2021-01-01:
Fans have some "inertia" when they are off, that is why you can pass high value with `--pwm-start-value` to ensure they start moving.

What is the value you currently using, and can you try with `--pwm-start-value 255`?

mindrunner, 2021-01-02:
Good point! I am also oberving, that sometimes only one of the two hdd fans is spinning. I will try out your suggestion! :+1: 
Cheers

mindrunner, 2021-01-11:
Turned out I was using a silencer adapter from Noctura which seem to not proper work with PWM. I removed the silencer, switched to the primary PWM controller and played around with the `hddfancontrol` parameters. Works way better now :)

```
HDDFANCONTROL_ARGS="-d /dev/sda /dev/sdb /dev/sdc /dev/sdd /dev/sde /dev/sdf -p /sys/class/hwmon/hwmon2/device/pwm2 /sys/class/hwmon/hwmon2/device/pwm3 --pwm-start-value 89 89 --pwm-stop-value 84 84 --min-fan-speed-prct 0 -i 60 --spin-down-time 1800 --min-temp 40 --max-temp 70"
```

closed 2021-01-11 by mindrunner