desbma/hddfancontrol #30: Hddfancontrol service start failure

Yoda-XY, 2021-03-13:
Mär 13 23:05:23 NAS-Server systemd[1]: hddfancontrol.service: Failed to load environment files: No such file or directory
Mär 13 23:05:23 NAS-Server systemd[1]: hddfancontrol.service: Failed to run 'start' task: No such file or directory
Mär 13 23:05:23 NAS-Server systemd[1]: hddfancontrol.service: Failed with result 'resources'.
Mär 13 23:05:23 NAS-Server systemd[1]: Failed to start HDD fan control.

no hddfancontrol file in /usr/bin.

daemon is working without start script

desbma, 2021-03-13:
How did you install hddfancontrol ?

Yoda-XY, 2021-03-13:
first from source than from pip.  the file doenst exist :/

desbma, 2021-03-13:
There is no systemd service if you installed from source or pip.

To install it:
* copy `systemd/hddfancontrol.service` from the repo to `/etc/systemd/system/hddfancontrol.service`
* change `/etc/systemd/system/hddfancontrol.service` to set the location of `hddfancontrol` (can be found with `which hddfancontrol`)
* copy `systemd/hddfancontrol.conf` from the repo to `/etc/conf.d/hddfancontrol` 
* set configuration in `/etc/conf.d/hddfancontrol`

Yoda-XY, 2021-03-13:
Thanks i changed the location to ExecStart=/usr/local/bin/hddfancontrol $HDDFANCONTROL_ARGS and reboot and start it again same error.... cant understand

desbma, 2021-03-14:
What is the exact error you are getting?

Yoda-XY, 2021-03-14:
Hello, thank you for your feedback! The start service script just doesn't work. I have now simply entered hddfancontrol in rc.local so that it now runs. I followed your instructions exactly. The path was really different for me. I have adjusted it. I also reloaded systemctl deamon. The error is displayed as above in the 1st post

desbma, 2021-03-14:
The error seem to indicate that `/etc/conf.d/hddfancontrol` is missing.

You can probably get more details by running `journalctl -f -u hddfancontrol` and `systemctl restart hddfancontrol` in another terminal.

Yoda-XY, 2021-03-14:
Hm the file exist and is in this location. Set to chmod 755

This is the file

#
# HDD fan control daemon startup arguments, for use as a systemd service
# This file should be located at /etc/conf.d/hddfancontrol
#
# Run 'hddfancontrol -h' to see help about all available arguments.
# Note that the parameters -b/--background, -l/--log-file and --pid-file should not be used here, as systemd handles both daemonization of the process and logging.
#

HDDFANCONTROL_ARGS="-d /dev/sda -p /sys/class/hwmon/hwmon2/pwm1 --pwm-start-value 200 --pwm-stop-value 45 --min-fan-speed-prct 0 --min-temp 35 --max-temp 40 -i 60"

desbma, 2021-03-14:
> Set to chmod 755

Not a good idea...

This issue has nothing to do with hddfancontrol, read the systemd logs, and it will tell you exactly what is wrong.

Yoda-XY, 2021-03-14:
Ok now i find the error its in

[Service]
EnvironmentFile=/etc/conf.d/hddfancontrol

it should be 

[Service]
EnvironmentFile=/etc/conf.d/hddfancontrol.conf

now it works! Thank you so much for your work!

closed 2021-03-14 by Yoda-XY