desbma/hddfancontrol #31: AUR upgrade to 1.4.0 failing

nightah, 2021-03-20:
Upon attempting an upgrade for `hddfancontrol` via the AUR I get the following when attempting to build the package during the `check()`:

```
==> Starting check()...
running test
WARNING: Testing via this command is deprecated and will be removed in a future version. Users looking for a generic test entry point independent of test runner are encouraged to use tox.
running egg_info
creating hddfancontrol.egg-info
writing hddfancontrol.egg-info/PKG-INFO
writing dependency_links to hddfancontrol.egg-info/dependency_links.txt
writing entry points to hddfancontrol.egg-info/entry_points.txt
writing requirements to hddfancontrol.egg-info/requires.txt
writing top-level names to hddfancontrol.egg-info/top_level.txt
writing manifest file 'hddfancontrol.egg-info/SOURCES.txt'
reading manifest file 'hddfancontrol.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'hddfancontrol.egg-info/SOURCES.txt'
running build_ext
test_compareActivityStats (tests.TestDrive)
Test drive stat analysis to detect activity. ... Drive does not support SCT temp query
ok
test_getActivityStats (tests.TestDrive)
Test drive stats fetching. ... Drive does not support SCT temp query
FAIL
test_getPrettyName (tests.TestDrive)
Test generation of pretty drive name. ... Drive does not support SCT temp query
ok
test_getState (tests.TestDrive)
Test drive state identification. ... Drive does not support SCT temp query
ok
test_getTemperature (tests.TestDrive)
Test device temperature probing. ... Drive does not support SCT temp query
Hddtemp daemon returned an error when querying temperature for this drive, falling back to hddtemp process invocation. If that happens often, you may need to raise the hddtemp daemon priority (see: https://github.com/desbma/hddfancontrol/issues/15#issuecomment-461405402).
ok
test_isSleeping (tests.TestDrive)
Test sleeping device identification. ... Drive does not support SCT temp query
ok
test_spinDown (tests.TestDrive)
Test HDD spin down. ... Drive does not support SCT temp query
ok
test_supportsHitachiTempQuery (tests.TestDrive)
Test detection for "Hitachi" temp query. ... Drive does not support SCT temp query
Drive does not support HGST temp query
Drive does not support HGST temp query
Drive does not support HGST temp query
ok
test_supportsSctTempQuery (tests.TestDrive)
Test detection for "SCT" temp query. ... Drive does not support SCT temp query
Drive does not support SCT temp query
ok

======================================================================
FAIL: test_getActivityStats (tests.TestDrive)
Test drive stats fetching.
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/amir/.cache/yay/hddfancontrol/src/hddfancontrol-1.4.0/tests/__init__.py", line 589, in test_getActivityStats
    self.drive.getActivityStats()
AssertionError: Exception not raised

----------------------------------------------------------------------
Ran 9 tests in 0.541s

FAILED (failures=1)
Test failed: <unittest.runner.TextTestResult run=9 errors=0 failures=1>
error: Test failed: <unittest.runner.TextTestResult run=9 errors=0 failures=1>
==> ERROR: A failure occurred in check().
    Aborting...
error making: hddfancontrol
```

The server in question has a mixture of SATA and Enterprise SAS drives.
This is also running Linux Kernel 5.11.7.

Please let me know if there's any other information that may be helpful to debugging this issue.

desbma, 2021-03-20:
Nothing wrong on your side, its a bug in the automated tests that assumed `/dev/sdz` does not exist.

desbma, 2021-03-20:
Should be fixed in 1.4.1.

nightah, 2021-03-21:
@desbma looks like the issue still persists with 1.4.1:

```
==> Starting check()...
running test
WARNING: Testing via this command is deprecated and will be removed in a future version. Users looking for a generic test entry point independent of test runner are encouraged to use tox.
running egg_info
creating hddfancontrol.egg-info
writing hddfancontrol.egg-info/PKG-INFO
writing dependency_links to hddfancontrol.egg-info/dependency_links.txt
writing entry points to hddfancontrol.egg-info/entry_points.txt
writing requirements to hddfancontrol.egg-info/requires.txt
writing top-level names to hddfancontrol.egg-info/top_level.txt
writing manifest file 'hddfancontrol.egg-info/SOURCES.txt'
reading manifest file 'hddfancontrol.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'hddfancontrol.egg-info/SOURCES.txt'
running build_ext
test_compareActivityStats (tests.TestDrive)
Test drive stat analysis to detect activity. ... Drive does not support SCT temp query
ok
test_getActivityStats (tests.TestDrive)
Test drive stats fetching. ... Drive does not support SCT temp query
FAIL
test_getPrettyName (tests.TestDrive)
Test generation of pretty drive name. ... Drive does not support SCT temp query
ok
test_getState (tests.TestDrive)
Test drive state identification. ... Drive does not support SCT temp query
ok
test_getTemperature (tests.TestDrive)
Test device temperature probing. ... Drive does not support SCT temp query
Hddtemp daemon returned an error when querying temperature for this drive, falling back to hddtemp process invocation. If that happens often, you may need to raise the hddtemp daemon priority (see: https://github.com/desbma/hddfancontrol/issues/15#issuecomment-461405402).
ok
test_isSleeping (tests.TestDrive)
Test sleeping device identification. ... Drive does not support SCT temp query
ok
test_spinDown (tests.TestDrive)
Test HDD spin down. ... Drive does not support SCT temp query
ok
test_supportsHitachiTempQuery (tests.TestDrive)
Test detection for "Hitachi" temp query. ... Drive does not support SCT temp query
Drive does not support HGST temp query
Drive does not support HGST temp query
Drive does not support HGST temp query
ok
test_supportsSctTempQuery (tests.TestDrive)
Test detection for "SCT" temp query. ... Drive does not support SCT temp query
Drive does not support SCT temp query
ok

======================================================================
FAIL: test_getActivityStats (tests.TestDrive)
Test drive stats fetching.
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/amir/.cache/yay/hddfancontrol/src/hddfancontrol-1.4.1/tests/__init__.py", line 589, in test_getActivityStats
    self.drive.getActivityStats()
AssertionError: Exception not raised

----------------------------------------------------------------------
Ran 9 tests in 0.541s

FAILED (failures=1)
Test failed: <unittest.runner.TextTestResult run=9 errors=0 failures=1>
error: Test failed: <unittest.runner.TextTestResult run=9 errors=0 failures=1>
==> ERROR: A failure occurred in check().
    Aborting...
error making: hddfancontrol
```

desbma, 2021-03-21:
It should **really** be fixed in 1.4.2 :)

closed 2021-03-21 by desbma

nightah, 2021-03-21:
Confirmed, thanks @desbma.