desbma/hddfancontrol #32: hddfancontrol crashes when Drive state: UNKNOWN

fightforlife, 2021-09-02:
I dont know if this is intended, but sometimes hddfancontrol crashes when drivestate is UNKNOWN. This leads to every timer resetting. 

DEBUG [sdc ST4000VN008-2DR166] Drive state: UNKNOWN
ERROR [Main] ValueError: invalid literal for int() with base 10: ''

`- ARGS=-d /dev/sdb1 /dev/sdc1 /dev/sdd1 -p /sys/class/hwmon/hwmon2/pwm1 /sys/class/hwmon/hwmon2/pwm2 --pwm-start-value 70 80 --pwm-stop-value 20 30 --min-temp 40 --max-temp 60 --min-fan-speed-prct 0 -i 60 --spin-down-time 900 -v debug -l /var/log/hddfancontrol.log`

```
DEBUG [sdd ST33000651AS] Drive state: ACTIVE_IDLE

DEBUG [DriveSpinDownThread-sdd ST33000651AS] Sleeping for 60 seconds

DEBUG [sdb WDC WD15EADS-00S2B0] Drive state: ACTIVE_IDLE

DEBUG [DriveSpinDownThread-sdb WDC WD15EADS-00S2B0] Sleeping for 60 seconds

DEBUG [DriveSpinDownThread-sdc ST4000VN008-2DR166] Drive is inactive for 4 min

DEBUG [sdc ST4000VN008-2DR166] Drive state: ACTIVE_IDLE

DEBUG [DriveSpinDownThread-sdc ST4000VN008-2DR166] Sleeping for 60 seconds

DEBUG [sdb WDC WD15EADS-00S2B0] Drive state: ACTIVE_IDLE

DEBUG [sdb WDC WD15EADS-00S2B0] Drive temperature: 43 °C

DEBUG [sdc ST4000VN008-2DR166] Drive state: ACTIVE_IDLE

DEBUG [sdc ST4000VN008-2DR166] Drive temperature: 32 °C

DEBUG [sdd ST33000651AS] Drive state: ACTIVE_IDLE

DEBUG [sdd ST33000651AS] Drive temperature: 41 °C

INFO [Main] Maximum device temperature: 43 °C

DEBUG [Main] Sleeping for 20 seconds

DEBUG [sdb WDC WD15EADS-00S2B0] Drive state: ACTIVE_IDLE

DEBUG [sdb WDC WD15EADS-00S2B0] Drive temperature: 43 °C

DEBUG [sdc ST4000VN008-2DR166] Drive state: ACTIVE_IDLE

DEBUG [sdc ST4000VN008-2DR166] Drive temperature: 32 °C

DEBUG [sdd ST33000651AS] Drive state: ACTIVE_IDLE

DEBUG [sdd ST33000651AS] Drive temperature: 41 °C

INFO [Main] Maximum device temperature: 43 °C

DEBUG [Main] Sleeping for 20 seconds

DEBUG [sdb WDC WD15EADS-00S2B0] Drive state: ACTIVE_IDLE

DEBUG [sdb WDC WD15EADS-00S2B0] Drive temperature: 43 °C

DEBUG [sdc ST4000VN008-2DR166] Drive state: ACTIVE_IDLE

DEBUG [sdc ST4000VN008-2DR166] Drive temperature: 32 °C

DEBUG [sdd ST33000651AS] Drive state: ACTIVE_IDLE

DEBUG [sdd ST33000651AS] Drive temperature: 41 °C

INFO [Main] Maximum device temperature: 43 °C

DEBUG [Main] Sleeping for 20 seconds

[... 311 lines cut ...]
DEBUG [sdc ST4000VN008-2DR166] Drive temperature: 32 °C

DEBUG [sdd ST33000651AS] Drive state: ACTIVE_IDLE

DEBUG [sdd ST33000651AS] Drive temperature: 41 °C

INFO [Main] Maximum device temperature: 43 °C

INFO [Fan #1] Setting fan speed to 15%

DEBUG [Fan #1] Rotation speed is currently 841 RPM

WARNING [Fan #1] /sys/class/hwmon/hwmon2/pwm1_enable was 0, setting it to 1

DEBUG [Fan #1] Setting PWM value to 55

INFO [Fan #2] Setting fan speed to 15%

DEBUG [Fan #2] Rotation speed is currently 1055 RPM

WARNING [Fan #2] /sys/class/hwmon/hwmon2/pwm2_enable was 0, setting it to 1

DEBUG [Fan #2] Setting PWM value to 63

DEBUG [Main] Sleeping for 20 seconds

DoXer67, 2021-09-02:
On which kernel?

fightforlife, 2021-09-02:
In docker on an host with:
 cat /proc/version
Linux version 5.11.0-31-generic (buildd@lcy01-amd64-009) (gcc (Ubuntu 10.3.0-1ubuntu1) 10.3.0, GNU ld (GNU Binutils for Ubuntu) 2.36.1) #33-Ubuntu SMP Wed Aug 11 13:19:04 UTC 2021

I know switched to --smartctl. Currently it is not hapnnening again.

desbma, 2021-09-06:
> DEBUG [sdc ST4000VN008-2DR166] Drive state: UNKNOWN
> ERROR [Main] ValueError: invalid literal for int() with base 10: ''

What happened is that `hdparm -C /dev/xxx` returned  "_unknown_" state for that drive, and the following temp probing failed too, which causes `hddfancontrol` to stop, and it was probably restarted automatically by your init system or something else.

I don't known what might have caused this. Since the drive state before the error was idle, maybe your drive has some non standard power saving feature that activates after a few minutes in idle? I know SATA ALPM is known to be buggy for some drives and sometimes the kernel blacklists known model that exhibits those issues. Do you have TLP running or some power saving tweak related to hard drives?

DoXer67, 2021-09-08:
I also had the UNKNOWN on two of my newer drives (WD 12TB). Maybe try with the newest kernel. I'm on Linux NAS 5.13.13-arch1-1 and until now I have no UNKNOWN from the drives.

fightforlife, 2021-09-10:
After switching to `--smartctl` it did not happen againg for around 5 days. Before it was happening everyday at least once.

fightforlife, 2022-01-02:
It has not happened again with the newest build and kernel in the last month. I will close this now.

closed 2022-01-02 by fightforlife