desbma/hddfancontrol #33: External command instead of PWM - feature request

nemihome, 2021-09-29:
Hi, I'm running an Odroid H2+ as NAS. Due to the fact that the PWM is only linked to the CPU I wrote a script to control the fan by CPU Temp and HDD temps. But then I realized that temperature reading with disk in active mode prevents spin down (in standby temp reading is working without spinning the disks up) - yes others realized this before which I realized after some google activity. ;)

Due to that I searched a program which could read temps without preventing spin down (I have not found any but I found hddfancontrol). The only problem is: My PWM fan can only be controlled via BIOS. There is no option to change the PWM outside BIOS - I already tried to change the settings but BIOS directly overwrites the values. So I bought a USB-Relay to do a very basic on / off control (this is ok, because the fans is running at a very low speed). The control of the relay is very simple - just usbrelay BITFT_1=0 (OFF) / usbrelay BITFT_1=1 (ON) in command line / bash.

The option to spin down drive e.g. after 10 minutes with temperature reading sounds interesting. But if I would use hddfanctontrol I would need an option to control the relay instead of PWM (kind of external command to switch on / off) or just to read temperatures to control the relay via my own script (in this case spin down would be done by hddfancontrol and usbrelay switching done by my script). For that I would need the temperate output per drive (optimum would be a verbose output when calling a command). This could be used for USB relay switching and fan control.

Is this something you would consider to be implemented?

desbma, 2021-10-02:
I am having a hard time to understand your verbose explanation (non english speaker here).

If you want to use your script for fan control, and only need hddfancontrol for auto spin down, maybe you should use [hd-idle](http://hd-idle.sourceforge.net/) instead, or its [similarly named recent rewrite](https://github.com/adelolmo/hd-idle).

I don't intend to add the ability to start an external script to control fans, that seems like a very rare/niche need. However if you want to use only hddfancontrol, and redirect to your script that should be doable without touching the hddfancontrol code. You just need to create files in the same directory (ie `/run/user/1000/my_fake_pwm_sysfs`) to simulate the PWM sysfs:
* `pwm1`: file only written by  `hddfancontrol` with value 0-255, can be a pipe that your script reads on the other end to control the fan
* `fan1_input`: used by `hddfancontrol` to read the fan RPM to know if the fan is moving, your script can write a value in there at every interval
* `fan1_enable`: read/written by `hddfancontrol` to instruct the kernel to take control of the PWM. Write `1` in it and `hddfancontrol` will not touch it.

Then pass `/run/user/1000/my_fake_pwm_sysfs/pwm1` as the PWM argument to hddfancontrol, and that should do the trick. Then you can automate the creation of the 3 files in a systemd unit, that the `hddfancontrol` service depends on, and that should work together nicely and reliably.

nemihome, 2021-10-04:
Hello, sorry for giving you a hard time. ;)

Spinning down the HDDs is not the problem but checking the temps when they are running without ressetting spindown timer is. Currently I'm just checking the following in my script: status HDD A, Status HDD B + temp of processor. If A or B is not standby or processor is above given limit the fan is running. Better would be if the HDDs are also checked for temperature without killing spindown timers.

I will try the variant you proposed. Thanks.

marki555, 2022-04-06:
@nemihome I have Odroid H2. I was able to control the fan, see some details here: https://stackoverflow.com/a/71758164/1129642
So the BIOS does not overwrite any values, it just configures the emc2103 chip to use its lookup tables (LUT) and then it ignores the direct RPM setting via `fan1_target`. So you can either disable LUT mode or customize it.
I will probably let the emc2103 control the fan and just push the HDD temp to it every few minutes. I log HDD temp to a file every 5 minutes by using `hdparm -H` command and HDD goes to sleep without issues (I am using `hd-idle` for that).

closed 2022-06-21 by desbma