desbma/hddfancontrol #34: Feature Request - Use drivetemp Kernel Module to Read Temps

Stonyx, 2021-11-08:
Last year the drivetemp kernel module was released as part of the Linux kernel to finally standardize reading hard drive temperatures.  It exposes hard drive temperatures via the hwmon functionality in the sysfs file system like every single other temperature sensor reading so getting the hard drive temperatures is now as simple as getting the CPU temperature.  Since this seems to be the recommended way of getting hard drive temperatures going forwards I would like to suggest that this ability be added to hddfancontrol.  Since hddfancontrol already supports reading CPU temperatures I'm hoping it will be fairly simple to implement the same functionality for hard drive temperatures.

desbma, 2021-11-08:
Yes, I will definitely look into that.

desbma, 2021-11-10:
I have added `drivetemp` support in the `drivetemp` branch: https://github.com/desbma/hddfancontrol/tree/drivetemp.
I am currently testing it on my own NAS, we'll see how that goes.
So far it looks good because temp queries do not affect drive stats, so all the workarounds of the other methods to make auto spin down work are not needed.

Stonyx, 2021-11-11:
Wow that was quick.  Awesome news.

hemagi, 2021-11-13:
I would want to test it, too.

I'm using hddfancontrol together with hdd-spindown.sh for years now. A few days ago I upgraded to bullseye, and the combination did not work for whatever reason.
Therefore I tried the spindown feature provided by hddfancontrol, without complete success.

So I hope that the drivetemp feature makes it possible to spindown the 7 snapraid drives in my server again

desbma, 2021-11-13:
> Therefore I tried the spindown feature provided by hddfancontrol, without complete success.
> So I hope that the drivetemp feature makes it possible to spindown the 7 snapraid drives in my server again

The workarounds done to fix https://github.com/desbma/hddfancontrol/issues/28 have helped, but with drivetemp it should be even more reliable. Not all drives support drivetemp though.

I have made some additional commits this morning in the `drivetemp` branch. My own drives are correctly spun down. If nothing catches fire, I will merge it and release version 1.5.0 tomorrow.

desbma, 2021-11-14:
`drivetemp` support merged, and version 1.5.0 released.
Just ensure the `drivetemp` kernel module is loaded (`modprobe -v drivetemp`) and hddfancontrol will use it if a drive supports it.

closed 2021-11-14 by desbma