desbma/hddfancontrol #35: Adaptec Aarcraid Possible HDD Temp?

Yoda-XY, 2022-01-21:
Hello how can i get the temperature behind my aptec raid controller. I have a raid 1. With this Smartctl command i can get the temperature but it doesnt work with hddtemp.
Thanks!!!

smartctl -A -d "aacraid,0,0,0" /dev/sda

In hddfancontrolconf ?

HDDFANCONTROL_ARGS="-d "aacraid,0,0,0" /dev/sda -p /sys/class/hwmon/hwmon2/pwm1 --$

desbma, 2022-01-21:
Hello,

Does your raid controller expose a block device ?

What is the output of `ls -l /dev/disk/by-id` ?

Yoda-XY, 2022-01-21:
Hello thank you for your help!

It show;

dev/disk/by-id
insgesamt 0
lrwxrwxrwx 1 root root  9 Jan 20 22:49 ata-SanDisk_SDSSDA120G_152975407983 -> ../../sdb
lrwxrwxrwx 1 root root 10 Jan 20 22:49 ata-SanDisk_SDSSDA120G_152975407983-part1 -> ../../sdb1
lrwxrwxrwx 1 root root 10 Jan 20 22:49 ata-SanDisk_SDSSDA120G_152975407983-part2 -> ../../sdb2
lrwxrwxrwx 1 root root 10 Jan 20 22:49 ata-SanDisk_SDSSDA120G_152975407983-part3 -> ../../sdb3
lrwxrwxrwx 1 root root  9 Jan 20 22:49 scsi-2ad07269100d00000 -> ../../sda
lrwxrwxrwx 1 root root  9 Jan 20 22:49 usb-USB_SanDisk_3.2Gen1_0101154d5e623fbf1eefa8b26751cc0cde2a30f25beadcd00edc62248d5cb69a6c4200000000000000000000448f0916000b240081558107a2295c89-0:0 -> ../../sdc
lrwxrwxrwx 1 root root  9 Jan 20 22:49 wwn-0x5001b44ea9a0776f -> ../../sdb
lrwxrwxrwx 1 root root 10 Jan 20 22:49 wwn-0x5001b44ea9a0776f-part1 -> ../../sdb1
lrwxrwxrwx 1 root root 10 Jan 20 22:49 wwn-0x5001b44ea9a0776f-part2 -> ../../sdb2
lrwxrwxrwx 1 root root 10 Jan 20 22:49 wwn-0x5001b44ea9a0776f-part3 -> ../../sdb3
root@NAS-Server ~ >

Yoda-XY, 2022-01-21:
And no i dont think so that the controller support block device. It cant support hba :/
So is that impossible to get the temperature via hddfancontrol with a aacraid command?

Yoda-XY, 2022-01-23:
so i had another idea and that is to integrate the command directly into your code but unfortunately this does not work either do you have an idea?

the direct command via smartctl 

`root@NAS-Server ~/hddfancontrol > smartctl -A -d aacraid,0,0,0 /dev/sda
smartctl 6.6 2017-11-05 r4594 [x86_64-linux-5.10.0-0.bpo.3-rt-amd64] (local build)
Copyright (C) 2002-17, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF READ SMART DATA SECTION ===
SMART Attributes Data Structure revision number: 16
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
  1 Raw_Read_Error_Rate     0x000b   100   100   001    Pre-fail  Always       -       0
  2 Throughput_Performance  0x0004   135   135   054    Old_age   Offline      -       108
  3 Spin_Up_Time            0x0007   081   081   001    Pre-fail  Always       -       389 (Average 391)
  4 Start_Stop_Count        0x0012   096   096   000    Old_age   Always       -       1700
  5 Reallocated_Sector_Ct   0x0033   100   100   001    Pre-fail  Always       -       0
  7 Seek_Error_Rate         0x000a   100   100   001    Old_age   Always       -       0
  8 Seek_Time_Performance   0x0004   133   133   020    Old_age   Offline      -       18
  9 Power_On_Hours          0x0012   099   099   000    Old_age   Always       -       8665
 10 Spin_Retry_Count        0x0012   100   100   001    Old_age   Always       -       0
 12 Power_Cycle_Count       0x0032   098   098   000    Old_age   Always       -       160
 22 Unknown_Attribute       0x0023   100   100   025    Pre-fail  Always       -       100
192 Power-Off_Retract_Count 0x0032   100   100   000    Old_age   Always       -       1902
193 Load_Cycle_Count        0x0012   100   100   000    Old_age   Always       -       1902
194 Temperature_Celsius     0x0002   037   037   000    Old_age   Always       -       41 (Min/Max 0/48)
196 Reallocated_Event_Count 0x0032   100   100   000    Old_age   Always       -       0
197 Current_Pending_Sector  0x0022   100   100   000    Old_age   Always       -       0
198 Offline_Uncorrectable   0x0008   100   100   000    Old_age   Offline      -       0
199 UDMA_CRC_Error_Count    0x000a   100   100   000    Old_age   Always       -       0

root@NAS-Server ~/hddfancontrol >`

`cmd = ("smartctl", "-A", "-d", "aacraid,0,0,0", self.device_filepath)`

and the config command is

`HDDFANCONTROL_ARGS="--smartctl -d /dev/sda -p /sys/class/hwmon/hwmon2/pwm1 --pwm- ...............................`

but it show

`Jan 23 13:18:04 NAS-Server systemd[1]: Started HDD fan control.
Jan 23 13:18:05 NAS-Server hddfancontrol[25835]: ERROR [Main] AssertionError:
Jan 23 13:18:05 NAS-Server hddfancontrol[25835]: INFO [Fan #1] Setting fan speed toJan 23 13:18:05 NAS-Server systemd[1]: hddfancontrol.service: Succeeded.
lines 1-10/10 (END)`

desbma, 2022-01-23:
Your controller seem to expose `/dev/sda` as a block device. I could add a custom parameter to allow passing `aacraid,0,0,0` to `smartctl`, but before doing that I need to sure the rest of the drive handling would work, since I can't test it myself. Unfortunately I think hdparm only works for SATA drives.

Does `hdparm -C /dev/sda` properly show the drive state?
Does `hdparm -y /dev/sda` triggers drive spin down?

Yoda-XY, 2022-01-23:
Hello thank you! No hdparm is not supported only the command in smartctl is working. Why do you need hdparm? The standby is handled by the controller itself.

Yoda-XY, 2022-01-24:
The topic can be closed because now also the standby function with smartctl can not be realized. The command -n standby does not work either. I will now buy an external temperature sensor, thank you!

closed 2022-01-24 by Yoda-XY