desbma/hddfancontrol #36: pwmX_enable is missing

jasonyzhang, 2022-04-26:
Hi,

I installed hddfancontrol using pip, and I am running the following command:
`sudo hddfancontrol -d /dev/sdb -p /sys/class/hwmon/hwmon4/pwm2 --pwm-start-value 200 --pwm-stop-value 75 --min-fan-speed-prct 20 -i 10 --smartctl`

The output I get is:
```
2022-04-25 23:41:04,454 INFO [Main] Process real time scheduler set to 2, priority 49
2022-04-25 23:41:04,459 INFO [sdb ST8000NE001-2M7101] Drive does not support native drivetemp temp query
2022-04-25 23:41:04,483 INFO [sdb ST8000NE001-2M7101] Will probe temperature with method SMARTCTL_SCT_INVOCATION
2022-04-25 23:41:04,505 INFO [Main] Maximum device temperature: 51 °C
2022-04-25 23:41:04,505 INFO [Fan #1] Setting fan speed to 100%
2022-04-25 23:41:04,505 ERROR [Main] FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/hwmon/hwmon4/pwm2_enable'
2022-04-25 23:41:04,505 INFO [Fan #1] Setting fan speed to 100%
Traceback (most recent call last):
  File "/usr/local/bin/hddfancontrol", line 8, in <module>
    sys.exit(cl_main())
  File "/usr/local/lib/python3.8/dist-packages/hddfancontrol/__init__.py", line 1306, in cl_main
    main(
  File "/usr/local/lib/python3.8/dist-packages/hddfancontrol/__init__.py", line 1089, in main
    fan.setSpeed(100)
  File "/usr/local/lib/python3.8/dist-packages/hddfancontrol/__init__.py", line 755, in setSpeed
    self.setPwmValue(target_value)
  File "/usr/local/lib/python3.8/dist-packages/hddfancontrol/__init__.py", line 760, in setPwmValue
    with open(self.enable_filepath, "r+t") as enable_file:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/hwmon/hwmon4/pwm2_enable'
```
When I `ls` inside of `/sys/class/hwmon/hwmon4`, I get:
```
fan1_input  fan1_label  fan2_input  fan2_label  name  power  pwm1  pwm2  subsystem  uevent
```

There indeed seems to be no `pwmX_enable`. Does this mean it is not possible to control my fan? This seems odd since `pwmconfig` is able to set my fan speed to 100%.

desbma, 2022-04-26:
Did you confirm this PWM can be controled by Linux?
For example does `echo 0 > /sys/class/hwmon/hwmon4/pwm2` stop the fan and `echo 255 > /sys/class/hwmon/hwmon4/pwm2` make it run at full speed?

jasonyzhang, 2022-04-26:
Actually, it seems that 0 doesn't quite turn off the fan but rather sets it to a much lower speed. Other numbers either set the fan to 100% or to the lower speed.

desbma, 2022-04-27:
It should be fixed by https://github.com/desbma/hddfancontrol/commit/47d9de32fed26829e46a8cc748173a3e4738f51d, can you try the `missing-enable-file` branch?

jasonyzhang, 2022-04-28:
Thanks! The patch works. 

An annoying behavior that I did notice is that setting the fan speed to the higher speed sometimes produces IO errors. I believe these errors to be spurious. This [thread ](https://github.com/lm-sensors/lm-sensors/pull/383) seems to document a similar issue. My machine is also a Dell Precision. When I call `echo 255 > /sys/class/hwmon/hwmon4/pwm2`, I often get `-bash: echo: write error: Input/output error` even though the fan speed changes accordingly.

This is caught by hddfancontrol (`ERROR [Main] OSError: [Errno 5] Input/output error`). For now, I am catching and ignoring OSErrors.

closed 2022-04-28 by jasonyzhang