desbma/hddfancontrol #38: template for hddfancontrol.conf

akrea, 2022-05-01:
Hi

This is not a real issue, but I didn't know how to contact you otherwise. In order to run the deamon I have to provide `HDDFANCONTROL_ARGS=""` in `hddfancontrol.conf`. Unfortunately I have no idea of the syntax to use (e.g. does every line need a dash in front of it or semicolon to end the line. or is it just on line of arguments like the hddfancontrol command). Maybe this is common knowledge among programmers and I'm just not there yet... If you could point me to the right documentation would be most appreciated.

Thank you!

desbma, 2022-05-01:
>  is it just on line of arguments like the hddfancontrol command

Yes it is the command line arguments.

akrea, 2022-05-02:
Does this look right (i have no idea how i could check that otherwise...).
```
HDDFANCONTROL_ARGS="-d /dev/sda /dev/sdb -p /sys/class/hwmon/hwmon1/pwm4 /sys/class/hwmon/hwmon1/pwm5 --pwm-start-value 200 200 --pwm-stop-value 45 60 --min-fan-speed-prct 10 -i 60 --spin-down-time 7200"
```

desbma, 2022-05-02:
Run it, and it will tell you if there is an incorrect argument.

akrea, 2022-05-02:
I think its working. Will closely monitor this next few days.
Thank you!!!

closed 2022-05-02 by desbma