desbma/hddfancontrol #40: ERROR [Main] KeyError: 'idle'

DoXer67, 2022-08-26:
Hi,
I use hddfancontrol with ARCH.

After installing a new TOSHIBA MG09ACA18TE I get the following error:

```
ERROR [DriveSpinDownThread-sdb TOSHIBA MG09ACA18TE] KeyError: 'idle'.
ERROR [Main] KeyError: 'idle'
``` 

Here is my config:
`HDDFANCONTROL_ARGS="-d /dev/sda /dev/sdb /dev/sdd /dev/sde /dev/sdf -p /sys/class/hwmon/hwmon10/pwm1 /sys/class/hwmon/hwmon10/pwm3 --min-temp 35 --max-temp 55 --pwm-start-value 150 150 --pwm-stop-value 100 100 --min-fan-speed-prct 0 -i 60 --spin-down-time 3600"`

/dev/sdb is the Toshiba 18GB                                  

What else is needed to isolate the error?

Here the complete log:
```
Aug 26 11:07:25 NAS hddfancontrol[37767]: INFO [sda WDC WD120EDAZ-11F3RA0] Will probe temperature with method DRIVETEM
Aug 26 11:07:25 NAS hddfancontrol[37767]: INFO [sdb TOSHIBA MG09ACA18TE] Will probe temperature with method DRIVETEMP 
Aug 26 11:07:25 NAS hddfancontrol[37767]: INFO [sdd ST18000NM000J-2TV103] Will probe temperature with method DRIVETEMP
Aug 26 11:07:25 NAS hddfancontrol[37767]: INFO [sde WDC WD120EDAZ-11F3RA0] Will probe temperature with method DRIVETEM
Aug 26 11:07:25 NAS hddfancontrol[37767]: INFO [sdf WDC WD160EDGZ-11B2DA0] Will probe temperature with method DRIVETEM
Aug 26 11:07:25 NAS hddfancontrol[37767]: ERROR [DriveSpinDownThread-sdb TOSHIBA MG09ACA18TE] KeyError: 'idle'        
Aug 26 11:07:25 NAS hddfancontrol[37767]: ERROR [Main] KeyError: 'idle'                                               
Aug 26 11:07:25 NAS hddfancontrol[37767]: INFO [Fan #1] Setting fan speed to 100%                                     
Aug 26 11:07:25 NAS hddfancontrol[37767]: INFO [DriveSpinDownThread-sda WDC WD120EDAZ-11F3RA0] Exiting                
Aug 26 11:07:25 NAS hddfancontrol[37767]: INFO [DriveSpinDownThread-sdd ST18000NM000J-2TV103] Exiting                 
Aug 26 11:07:25 NAS hddfancontrol[37767]: INFO [DriveSpinDownThread-sdf WDC WD160EDGZ-11B2DA0] Exiting                
Aug 26 11:07:25 NAS hddfancontrol[37767]: INFO [DriveSpinDownThread-sde WDC WD120EDAZ-11F3RA0] Exiting                
Aug 26 11:07:25 NAS hddfancontrol[37767]: WARNING [Fan #1] /sys/class/hwmon/hwmon10/pwm1_enable was 0, setting it to 1
Aug 26 11:07:25 NAS hddfancontrol[37767]: INFO [Fan #2] Setting fan speed to 100%                                     
Aug 26 11:07:25 NAS hddfancontrol[37767]: WARNING [Fan #2] /sys/class/hwmon/hwmon10/pwm3_enable was 0, setting it to 1
```

desbma, 2022-08-26:
Can you try commit 4e13dc67a062ef9eb193ee367dea706e60ae38b5?

DoXer67, 2022-08-26:
Sorry, I'm not a programmer. I understand to add the two lines. But where? Is the file on my system? Do I have to patch the file in the TAR-GZ (ARCH-Package)  and then reinstall?

desbma, 2022-08-26:
Do this as root:
```
cp /usr/lib/python3.10/site-packages/hddfancontrol/__init__.py{,.orig}
curl https://raw.githubusercontent.com/desbma/hddfancontrol/4e13dc67a062ef9eb193ee367dea706e60ae38b5/hddfancontrol/__init__.py > /usr/lib/python3.10/site-packages/hddfancontrol/__init__.py
systemctl restart hddfancontrol
```

DoXer67, 2022-08-26:
I added a
`rm /usr/lib/python3.10/site-packages/hddfancontrol/__init__.py`
after the copy-command. ;)

Your patch solved the problem. Looks good. Thank you for the fast response.

desbma, 2022-08-26:
> I added a `rm /usr/lib/python3.10/site-packages/hddfancontrol/__init__.py` after the copy-command. ;)
>

This is redundant and not needed.

> Your patch solved the problem. Looks good. Thank you for the fast response.

Great

closed 2022-08-26 by desbma