desbma/hddfancontrol #51: Allow use of '*' in config file for systemd service

megatog615, 2024-03-26:
Not sure if it affects the OpenRC service file but in my system if I specify `-d /dev/disk/by-path/*-sas-phy*` as part of my HDDFANCONTROL_ARGS in the configuration file for use with the systemd service file, it does not expand the asterisks like it does if I start hddfancontrol manually from the command line.

I am trying to pass every drive that is connected to my SAS controller as drives to monitor.

desbma, 2024-03-27:
Your shell does the glob extension, systemd does not do any.

If you want to use glob patterns, you have to modify [this line](https://github.com/desbma/hddfancontrol/blob/02e83f7455c63e68bb7213df4169693d01e8a637/systemd/hddfancontrol.service#L7) to for example:
```
ExecStart=/bin/bash -c 'hddfancontrol $HDDFANCONTROL_ARGS'
```
Preferably with a fragment file in `/etc/systemd/system/service.d/` (you'd need to add a `ExecStart=` before to avoid having 2 commands) so that it does not get overwritten when the package is updated.

megatog615, 2024-03-27:
I will try this, but is there some reason why this can't be included by default?

closed 2024-03-28 by desbma