desbma/hddfancontrol #52: Is there a way to get hddfancontrol to work with pipx?

TomTheDragon, 2024-04-24:
I recently upgraded a small server to Debian 12 and i noticed that hddfancontrol stopped working, since they recommend you using system packages or isolated installs over pipx "if you can" now days.
So i removed all of the remains of the old installation and i installed it back over pipx.
I also made sure that everything works by running hddfancontrol directly and i also made sure that the new path was updated in the service file.
I however noticed that it just errors out when i try to start it as a service and there was just nothing in the service log (journalctl).

desbma, 2024-04-25:
> I however noticed that it just errors out when i try to start it as a service and there was just nothing in the service log (journalctl).

How do you know it errors out if you have nothing in the systemd journal?

TomTheDragon, 2024-05-23:
I do get an error when i check the status.
```
root@omv:~# systemctl status hddfancontrol.service
× hddfancontrol.service - HDD fan control
     Loaded: loaded (/etc/systemd/system/hddfancontrol.service; disabled; preset: enabled)
     Active: failed (Result: exit-code) since Fri 2024-05-24 01:33:12 CEST; 2s ago
   Duration: 65ms
    Process: 155724 ExecStart=/root/.local/bin/hddfancontrol $HDDFANCONTROL_ARGS (code=exited, status=203/EXEC)
   Main PID: 155724 (code=exited, status=203/EXEC)
        CPU: 65ms
```

I am using the default service file and i only changed the location to the binary because of pipx.

It does however run if i run it directly like this:
```
/root/.local/bin/hddfancontrol -d /dev/sda -p /sys/class/hwmon/hwmon2/pwm2 --pwm-start-value 80 --pwm-stop-value 60 -i 60 --min-temp 35 --max-temp 40 --min-fan-speed-prct 25
```

desbma, 2024-05-24:
Try commenting out `ProtectHome=true` in `/etc/systemd/system/hddfancontrol.service`.

TomTheDragon, 2024-05-24:
That fixed it.
Thanks!

closed 2024-05-24 by desbma