desbma/hddfancontrol #55: Feature Request: Like CPU temperature it should be possible to add another sensors like System-Temperature

NilsRo, 2024-07-14:
Hi,

I am converting a Netgear ReadyNAS Ultra6 to OMV and like to control the single big fan with hddfancontrol. It has a passiv cooled Atom D510 CPU. I added the CPU to the hddfancontrol configuration and it works great. Now I also like to take the system temperature into account and from studying the docs it is actually not implemented. Is that right? 

Feature Request: It would be helpful to add also System Temperature for monitoring, at best adding additional sensors should not be limited.

Cheers,
Nils

desbma, 2024-07-14:
If you set the "system" sensor with `--cpu-sensor`, does that work as intended?
The best solution is probably to support an arbitrary number of hwmon sensors, CPU, system or anything.

NilsRo, 2024-07-14:
Yes, it works fine. It is simply another virtual file provided by hwmon. 

That would be nice especially for small NAS devices with one fan only.

desbma, 2024-07-14:
All right, I'll add support for an unlimited number of "non HDD" sensors then.

I'm currently in the final stage of a complete rewrite (testing it on my own NAS), so this will go in the [v2 branch](https://github.com/desbma/hddfancontrol/tree/v2).

FoxxMD, 2024-09-26:
Related to arbitrary sensors...I've got my NAS running in a proxmox VM and had to pass the whole SATA controller to the VM to make it work. Now all my HDD sensors are in the VM but fan sensors are still on the host. Does anyone know a way I can "fake" the HDD sensors on the host?

closed 2024-10-21 by desbma