desbma/hddfancontrol #57: hddfancontrol fails to build with setuptools 74+

rosset, 2024-10-19:
hddfancontrol fails to build with setuptools 74+

please see the details here: https://bugzilla.redhat.com/show_bug.cgi?id=2319631
builds logs available here:  https://download.copr.fedorainfracloud.org/results/churchyard/setuptools-74:custom:isolated/fedora-rawhide-x86_64/08147091-hddfancontrol/

thanks!

desbma, 2024-10-20:
Are you the maintainer of that package? 
A bit of analysis would not hurt before dropping an issue upstream with a dozen log files. :roll_eyes: 

I just tested and it builds fine on my system with setuptools 75. 
It seems to me the issue is on your (packaging) end, with the command you use to run tests (`python setup.py test`).
See for example the AUR package: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=hddfancontrol#n20

rosset, 2024-10-20:
> Are you the maintainer of that package? A bit of analysis would not hurt before dropping an issue upstream with a dozen log files. 🙄
> 
> I just tested and it builds fine on my system with setuptools 75. It seems to me the issue is on your (packaging) end, with the command you use to run tests (`python setup.py test`). See for example the AUR package: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=hddfancontrol#n20

hi @desbma  you are right, I was travelling and did not looked at the logs, it's a downstream issue, I'll fix during the week, thanks!

closed 2024-10-20 by rosset