desbma/hddfancontrol #58: "Unexpected mode: 5" on NCT6796D with "Full Speed" BIOS setting

h0m3us3r, 2024-10-29:


desbma, 2024-10-31:
Thanks for the report, I was not aware that the `enable` file may support driver specific values.

For reference, this is the kernel code for the supported vales in the driver in question: https://elixir.bootlin.com/linux/v6.11.5/source/drivers/hwmon/nct6775.h#L9

Rather that trying to support all possible values for present and future drivers, I have taken another approach than your PR, by ignoring the meaning for other values, but keeping the raw value if we need to restore it.

See https://github.com/desbma/hddfancontrol/commit/890a98f35694c4be25bbd8de2e0572ba3160fa3c

This should fix it

closed 2024-10-31 by desbma

h0m3us3r, 2024-10-31:
Yeah, your approach in https://github.com/desbma/hddfancontrol/commit/890a98f35694c4be25bbd8de2e0572ba3160fa3c is much better now that I think about it, thank you!