desbma/hddfancontrol #60: Provide prebuilt executables

kattjevfel, 2024-11-10:
Could prebuilt executables be added to the releases? This could be useful for users of some distros where the necessary build tools aren't available nor a packaged version.

This is the case with Debian stable, where it fails with the following:
> error: package `typed_floats_macros v1.0.2` cannot be built because it requires rustc 1.70 or newer, while the currently active rustc version is 1.63.0

There is no backport available and building it on another system gives glibc errors errors instead, leaving me unable to use the project.

I see github workflows is already being used for CI, perhaps this could be used to publish builds as well?

desbma, 2024-11-10:
Yes, I'll add Debian package generation in the CI pipeline for the next beta.

In the meantime, if you want to test the project, you can install a recent Rust toolchain with [rustup](https://rustup.rs/), and follow the [instructions to install from source](https://github.com/desbma/hddfancontrol?tab=readme-ov-file#from-source).
Debian being Debian, Rustup itself is not yet packaged in Debian stable repos (but it is in unstable).

desbma, 2024-11-10:
There you go https://github.com/desbma/hddfancontrol/releases

kattjevfel, 2024-11-11:
Thanks, that works perfectly! :)

judahrand, 2024-11-19:
@desbma There are cases where having a binary is also useful. For example on TrueNAS `apt` is disabled. It'd be great to have a binary which could be manually installed (could even be wrapped in a container image). Would it be possible to also supply the binaries with releases?

desbma, 2024-12-12:
Version 2.0.0-b2 now has amd64 static binary published on GitHub.
Open new issues if you need other architectures.

closed 2024-12-12 by desbma