desbma/hddfancontrol #61: Check other fan inputs for change in fan speed on a specific pwm control

RodoMa92, 2024-11-23:
The default logic in this software is to map pwmx to fanx_input always, but on a lot of motherboards this assumption breaks, like in my case (pwm3 is mapped to fan1_input). 

I've tried to add this myself into it, but I'm not skilled enough in rust to be able to do it myself. 

This might be accomplished in two ways AFAICS:

1) Allow to specify both pwm input and fan input to the daemon
2) Handle this dynamically in the code (which was the approach that I've tried to implement in my case)

desbma, 2024-11-23:
Can you post the output of `ls -l XXX` where XXX is the pwm dir?

It seems [pwmconfig from lm-sensors](https://github.com/lm-sensors/lm-sensors/tree/master) also relies on this assumption, can you confirm?

By the way `fanX_input` is only used to detect if the fan is moving when using `hddfancontrol pwm-test`, so the daemon operation is unaffected by this.

RodoMa92, 2024-11-23:
> It seems [pwmconfig from lm-sensors](https://github.com/lm-sensors/lm-sensors/tree/master) also relies on this assumption, can you confirm?
> 
> Can you post the output of `ls -l XXX` where XXX is the pwm dir?
> 
> By the way `fanX_input` is only used to detect if the fan is moving when using `hddfancontrol pwm-test`, so the daemon operation is unaffected by this.

pwmconfig do not assume the same, since it actually works with my board, see log below:
```
# pwmconfig version 3.6.0
This program will search your sensors for pulse width modulation (pwm)
controls, and test each one to see if it controls a fan on
your motherboard. Note that many motherboards do not have pwm
circuitry installed, even if your sensor chip supports pwm.

We will attempt to briefly stop each fan using the pwm controls.
The program will attempt to restore each fan to full speed
after testing. However, it is ** very important ** that you
physically verify that the fans have been to full speed
after the program has completed.

Found the following devices:
   hwmon0 is it8728
   hwmon1 is drivetemp
   hwmon2 is drivetemp
   hwmon3 is drivetemp
   hwmon4 is drivetemp
   hwmon5 is drivetemp
   hwmon6 is drivetemp
   hwmon7 is k10temp
   hwmon8 is fam15h_power
   hwmon9 is nouveau

Found the following PWM controls:
   hwmon0/pwm1           current value: 0
   hwmon0/pwm2           current value: 70
hwmon0/pwm2 is currently setup for automatic speed control.
In general, automatic mode is preferred over manual mode, as
it is more efficient and it reacts faster. Are you sure that
you want to setup this output for manual control? (n) y
   hwmon0/pwm3           current value: 255
   hwmon9/pwm1           current value: 40
hwmon9/pwm1 is currently setup for automatic speed control.
In general, automatic mode is preferred over manual mode, as
it is more efficient and it reacts faster. Are you sure that
you want to setup this output for manual control? (n) y

Giving the fans some time to reach full speed...
Found the following fan sensors:
   hwmon0/fan1_input     current speed: 1844 RPM
   hwmon0/fan2_input     current speed: 0 ... skipping!
   hwmon0/fan3_input     current speed: 0 ... skipping!
   hwmon0/fan4_input     current speed: 0 ... skipping!
   hwmon0/fan5_input     current speed: 1562 RPM

Warning!!! This program will stop your fans, one at a time,
for approximately 5 seconds each!!!
This may cause your processor temperature to rise!!!
If you do not want to do this hit control-C now!!!
Hit return to continue: 

Testing pwm control hwmon0/pwm1 ...
  hwmon0/fan1_input ... speed was 1844 now 1844
    no correlation
  hwmon0/fan5_input ... speed was 1562 now 1562
    no correlation

No correlations were detected.
There is either no fan connected to the output of hwmon0/pwm1,
or the connected fan has no rpm-signal connected to one of
the tested fan sensors. (Note: not all motherboards have
the pwm outputs connected to the fan connectors,
check out the hardware database on http://www.almico.com/forumindex.php)

Did you see/hear a fan stopping during the above test (n)? 

Testing pwm control hwmon0/pwm2 ...
  hwmon0/fan1_input ... speed was 1844 now 1844
    no correlation
  hwmon0/fan5_input ... speed was 1562 now 1562
    no correlation

No correlations were detected.
There is either no fan connected to the output of hwmon0/pwm2,
or the connected fan has no rpm-signal connected to one of
the tested fan sensors. (Note: not all motherboards have
the pwm outputs connected to the fan connectors,
check out the hardware database on http://www.almico.com/forumindex.php)

Did you see/hear a fan stopping during the above test (n)? 

Testing pwm control hwmon0/pwm3 ...
  hwmon0/fan1_input ... speed was 1844 now 346
    It appears that fan hwmon0/fan1_input
    is controlled by pwm hwmon0/pwm3
Would you like to generate a detailed correlation (y)? 
    PWM 255 FAN 1814
    PWM 240 FAN 1771
    PWM 225 FAN 1730
    PWM 210 FAN 1662
    PWM 195 FAN 1584
    PWM 180 FAN 1506
    PWM 165 FAN 1427
    PWM 150 FAN 1341
    PWM 135 FAN 1259
    PWM 120 FAN 1192
    PWM 105 FAN 1108
    PWM 90 FAN 1010
    PWM 75 FAN 909
    PWM 60 FAN 797
    PWM 45 FAN 676
    PWM 30 FAN 549
    PWM 28 FAN 475
    PWM 26 FAN 428
    PWM 24 FAN 360
    PWM 22 FAN 309
    PWM 20 FAN 279
    PWM 18 FAN 254
    PWM 16 FAN 185
    PWM 14 FAN 129
    PWM 12 FAN 90
    PWM 10 FAN 59
    PWM 8 FAN 0
    Fan Stopped at PWM = 8

  hwmon0/fan5_input ... speed was 1562 now 1555
    no correlation

Testing pwm control hwmon9/pwm1 ...
  hwmon0/fan1_input ... speed was 1844 now 1588
    no correlation
  hwmon0/fan5_input ... speed was 1562 now 1562
    no correlation

No correlations were detected.
There is either no fan connected to the output of hwmon9/pwm1,
or the connected fan has no rpm-signal connected to one of
the tested fan sensors. (Note: not all motherboards have
the pwm outputs connected to the fan connectors,
check out the hardware database on http://www.almico.com/forumindex.php)

Did you see/hear a fan stopping during the above test (n)? 

Testing is complete.
Please verify that all fans have returned to their normal speed.
```
Good to know that the only issue is the test then, I didn't knew. I just assumed it just would not control my fans at all. I'll test it shortly then :)

RodoMa92, 2024-11-23:
Yeah, can confirm that the only issue is the test. Software still controls the fans just fine, since fan speed is not a monitored parameter from the daemon. Thanks a lot for this software :) I'll probably leave this open for now just for the issue in the testing part :P

desbma, 2024-12-12:
This should be fixed in `2.0.0b2`: the `fanX_input` path is now resolved dynamically.

closed 2024-12-12 by desbma

RodoMa92, 2024-12-12:
Looks fine, I'll test it shortly and give feedback, but to me it looks properly handled now. Thanks a lot!