desbma/hddfancontrol #63: HD Idle with v2

abovethecloud, 2024-12-10:
I am still using the python version, which supports automatically spinning down drives.
v2 supports smartctl to query temperature, but reccommends using hd-idle to spin down drives, which states: "Important note: hd-idle is not compatible with the usage of disk monitoring tools like smartmontools."

What kind of temperature querying do you suggest to query drives temperature on debian?

desbma, 2024-12-10:
I don't think there are any issues when using hd-idle in combination with hddfancontrol when it internally uses smartctl (which by the way provides several different way to query temperature) to query temperature. 

I believe the warning on hd-idle README refers to using the periodic SMART tests, or periodic health monitoring feature from smartmontools. If the drive is spun down by hd-idle, and smartmontools tries to run a test, that might not go well, either because the test will error, or because the drive will be spun down and immediately up repeatedly which is not good for the drive either.

> What kind of temperature querying do you suggest to query drives temperature on debian?

Version 2 of hddfancontrol should pick the best version available on your system and supported by the drive, and handle its specifics transparently.

abovethecloud, 2024-12-11:
Thanks for the detailed and informative response. Out of curiosity, what is the order "best to worst" of methods to check for temperature according to the tool? Is it possible to check what method is actually used?

desbma, 2024-12-11:
> Out of curiosity, what is the order "best to worst" of methods to check for temperature according to the tool?

https://github.com/desbma/hddfancontrol/blob/96922572e68efcdc9058686faf91690c991859be/src/probe/mod.rs#L50-L57

> Is it possible to check what method is actually used?

Currently only the methods  tested and **not** supported are logged, I should probably improve that.

desbma, 2024-12-12:
The temperature method automatically chosen is now logged, for example:
```
INFO  [hddfancontrol::probe] Using probing method 'native Linux drivetemp' for drive 'sda WDC WD82PURZ-85TEUY0'
```

closed 2024-12-12 by desbma