desbma/hddfancontrol #64: "/sys/block/sdb/../../hwmon" does not exist

amo13, 2025-01-01:
Happy new year!

---------------

I see the following in my logs:

```
INFO  [hddfancontrol::probe] Drive 'sdd WDC WD60EFAX-68JH4N1' does not support probing method 'native Linux drivetemp': "/sys/block/sdd/../../hwmon" does not exist
INFO  [hddfancontrol::probe] Using probing method 'hdparm Hitachi/HGST' for drive 'sdd WDC WD60EFAX-68JH4N1'
INFO  [hddfancontrol::probe] Drive 'sdb WDC WD60EFRX-68L0BN1' does not support probing method 'native Linux drivetemp': "/sys/block/sdb/../../hwmon" does not exist
INFO  [hddfancontrol::probe] Using probing method 'hdparm Hitachi/HGST' for drive 'sdb WDC WD60EFRX-68L0BN1'
INFO  [hddfancontrol::probe] Drive 'sde WDC WD30EFRX-68EUZN0' does not support probing method 'native Linux drivetemp': "/sys/block/sde/../../hwmon" does not exist
INFO  [hddfancontrol::probe] Using probing method 'hdparm Hitachi/HGST' for drive 'sde WDC WD30EFRX-68EUZN0'
INFO  [hddfancontrol] Max drive temperature: -18°C
INFO  [hddfancontrol::fan] PWM nct6775.656/1 mode set from OFF to SOFTWARE
INFO  [hddfancontrol::fan] Fan nct6775.656/1 speed set to 20.0%
INFO  [hddfancontrol] Max drive temperature: -18°C
```

I am using this hddfancontrol 2.0.0-beta.3 on archlinux. Maybe there is something wrong with the paths?

Partial `tree` of `/sys/class`:
```
/sys/class/
├── hwmon
│   ├── hwmon2 -> ../../devices/platform/nct6775.656/hwmon/hwmon2
│   ├── hwmon1 -> ../../devices/platform/coretemp.0/hwmon/hwmon1
│   └── hwmon0 -> ../../devices/virtual/thermal/thermal_zone0/hwmon0
├── block
│   ├── sdg -> ../../devices/pci0000:00/0000:00:15.0/usb1/1-6/1-6:1.0/host13/target13:0:0/13:0:0:0/block/sdg
│   ├── sdf2 -> ../../devices/pci0000:00/0000:00:15.0/usb1/1-5/1-5:1.0/host12/target12:0:0/12:0:0:0/block/sdf/sdf2
│   ├── sdf1 -> ../../devices/pci0000:00/0000:00:15.0/usb1/1-5/1-5:1.0/host12/target12:0:0/12:0:0:0/block/sdf/sdf1
│   ├── sdf -> ../../devices/pci0000:00/0000:00:15.0/usb1/1-5/1-5:1.0/host12/target12:0:0/12:0:0:0/block/sdf
│   ├── sde -> ../../devices/pci0000:00/0000:00:13.3/0000:04:00.0/ata6/host5/target5:0:0/5:0:0:0/block/sde
│   ├── sdd -> ../../devices/pci0000:00/0000:00:13.3/0000:04:00.0/ata5/host4/target4:0:0/4:0:0:0/block/sdd
│   ├── sdc -> ../../devices/pci0000:00/0000:00:13.2/0000:03:00.0/ata4/host3/target3:0:0/3:0:0:0/block/sdc
│   ├── sdb -> ../../devices/pci0000:00/0000:00:12.0/ata2/host1/target1:0:0/1:0:0:0/block/sdb
│   ├── sda -> ../../devices/pci0000:00/0000:00:12.0/ata1/host0/target0:0:0/0:0:0:0/block/sda
│   ├── dm-4 -> ../../devices/virtual/block/dm-4
│   ├── dm-3 -> ../../devices/virtual/block/dm-3
│   ├── dm-2 -> ../../devices/virtual/block/dm-2
│   ├── dm-1 -> ../../devices/virtual/block/dm-1
│   └── dm-0 -> ../../devices/virtual/block/dm-0
```

Also, the WesternDigital Red drives I set to monitor appear to be wrongly recognized as Hitachi HGST drives.

Any help would be greatly appreciated.

desbma, 2025-01-01:
This should be fixed, you can check by compiling [from source](https://github.com/desbma/hddfancontrol?tab=readme-ov-file#from-source) if you like,  otherwise you'll have to wait for the next beta.

closed 2025-01-01 by desbma

desbma, 2025-01-02:
Version 2.0.0 beta4 is out with the fix: https://github.com/desbma/hddfancontrol/releases/tag/2.0.0.b4

amo13, 2025-01-02:
That was fast!

But I am getting the following error:
```
Jan 02 15:06:57 server systemd[1]: Started HDD fan control.
Jan 02 15:06:57 server (ncontrol)[236431]: hddfancontrol.service: Referenced but unset environment variable evaluates to an empty string: HDDFANCONTROL_LOG_LEVEL
Jan 02 15:06:57 server hddfancontrol[236431]: error: unexpected argument '-d' found
Jan 02 15:06:57 server hddfancontrol[236431]: Usage: hddfancontrol [OPTIONS] <COMMAND>
Jan 02 15:06:57 server hddfancontrol[236431]: For more information, try '--help'.
Jan 02 15:06:57 server systemd[1]: hddfancontrol.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Jan 02 15:06:57 server systemd[1]: hddfancontrol.service: Failed with result 'exit-code'.
Jan 02 15:06:57 server systemd[1]: hddfancontrol.service: Triggering OnFailure= dependencies.
```

desbma, 2025-01-02:
You need to update `/etc/conf.d/hddfancontrol`, `HDDFANCONTROL_LOG_LEVEL` was added, see https://github.com/desbma/hddfancontrol/blob/master/systemd/hddfancontrol.conf

amo13, 2025-01-02:
Oh yes. Sorry, I could have thought about this.
Works now, thank you so much!

-----

```
Jan 02 15:12:57 server systemd[1]: Started HDD fan control.
Jan 02 15:12:57 server hddfancontrol[237083]: INFO  [hddfancontrol::probe] Drive 'sdd WDC WD60EFAX-68JH4N1' does not support probing method 'native Linux drivetemp': "/sys/block/sdd/../../hwmon" does not exist
Jan 02 15:12:57 server hddfancontrol[237083]: INFO  [hddfancontrol::probe] Drive 'sdd WDC WD60EFAX-68JH4N1' does not support probing method 'hdparm Hitachi/HGST': hdparm returned soft error
Jan 02 15:12:57 server hddfancontrol[237083]: INFO  [hddfancontrol::probe] Using probing method 'smartctl SCT' for drive 'sdd WDC WD60EFAX-68JH4N1'
Jan 02 15:12:57 server hddfancontrol[237083]: INFO  [hddfancontrol::probe] Drive 'sdb WDC WD60EFRX-68L0BN1' does not support probing method 'native Linux drivetemp': "/sys/block/sdb/../../hwmon" does not exist
Jan 02 15:12:58 server hddfancontrol[237083]: INFO  [hddfancontrol::probe] Drive 'sdb WDC WD60EFRX-68L0BN1' does not support probing method 'hdparm Hitachi/HGST': hdparm returned soft error
Jan 02 15:12:58 server hddfancontrol[237083]: INFO  [hddfancontrol::probe] Using probing method 'smartctl SCT' for drive 'sdb WDC WD60EFRX-68L0BN1'
Jan 02 15:12:58 server hddfancontrol[237083]: INFO  [hddfancontrol::probe] Drive 'sde WDC WD30EFRX-68EUZN0' does not support probing method 'native Linux drivetemp': "/sys/block/sde/../../hwmon" does not exist
Jan 02 15:12:58 server hddfancontrol[237083]: INFO  [hddfancontrol::probe] Drive 'sde WDC WD30EFRX-68EUZN0' does not support probing method 'hdparm Hitachi/HGST': hdparm returned soft error
Jan 02 15:12:58 server hddfancontrol[237083]: INFO  [hddfancontrol::probe] Using probing method 'smartctl SCT' for drive 'sde WDC WD30EFRX-68EUZN0'
Jan 02 15:12:58 server hddfancontrol[237083]: INFO  [hddfancontrol] Max drive temperature: 36°C
Jan 02 15:12:58 server hddfancontrol[237083]: INFO  [hddfancontrol::fan] PWM nct6775.656/1 mode set from OFF to SOFTWARE
Jan 02 15:12:58 server hddfancontrol[237083]: INFO  [hddfancontrol::fan] Fan nct6775.656/1 speed set to 30.0%
```

Are you sure that `/sys/block/sdb/../../hwmon` is a correct path for looking for hwmon?

desbma, 2025-01-02:
Different temperature probing methods are tried in order, until one that works for the drive is found ("smartctl SCT" in your case).

This line:
`Drive 'sdd WDC WD60EFAX-68JH4N1' does not support probing method 'native Linux drivetemp': "/sys/block/sdd/../../hwmon" does not exist`
simply means that the drive does not support the "native Linux drivetemp" probing method. 
There is nothing wrong with that.

desbma, 2025-01-02:
To clarify: `/sys/block/sdX` are symlinks to much deeper paths, so for example, for a drive that supports drivetemp, `/sys/block/sdb/../../hwmon` resolves to something like `/sys/devices/pci0000:00/0000:00:1f.2/ata2/host1/target1:0:0/1:0:0:0/hwmon`.

amo13, 2025-01-02:
Yes, this is why I added the partial `tree` in my report: There *is* a `hwmon` in some deeper path on my system. Shouldn't hddfancontrol be able to find it instead of reporting that it is not there?

desbma, 2025-01-02:
So because your system has temperature sensors unrelated to the drive, hddfancontrol should randomly use it?
That would be going against precisely what this program is supposed to do.

amo13, 2025-01-02:
Ok, thanks for clarifying :)

commit 9a737319 (2025-01-01): fix: hdparm prober error handling (fix #64)