desbma/hddfancontrol #65: Using wildcard in pwm filepath

chbgdn, 2025-01-07:
After upgrading to hddfancontrol v2 there some kind of regression. I'm using drivetemp kernel driver to probing hdd temperatures, and seems it loads earlier than my driver mothrboard chip driver, so my pwm path looks like `/sys/devices/platform/it87.2608/hwmon/hwmon11/pwm3` and hwmon id depends on count of hdd connected or any other hwmon drivers loaded. In v1 I used wildcard in pwm path `/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3`, but this trick not working in v2 anymore. Can it be fixed?
```
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 10ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 15ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 22.5ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 33.75ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 50.625ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 75.9375ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 113.90625ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 170.859375ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 256.289062ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 384.433593ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 576.650389ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 864.975583ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 1s
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 1s
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 1s
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 1s
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 1s
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 1s
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 1s
Error: No such file or directory (os error 2)
```

desbma, 2025-01-07:
> In v1 I used wildcard in pwm path /sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3, but this trick not working in v2 anymore. 

What "trick" did you do, and how exactly ?
The log you have attached, shows the literal string `/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3` has been passed to hddfancontrol, and of course no file with that path will ever exist.

chbgdn, 2025-01-07:
`it87` kernel driver provides only one hwmon, but its identifier may change after reboot.
```
nas:~# ls -l /sys/devices/platform/it87.2608/hwmon/
total 0
drwxr-xr-x 3 root root 0 Dec 26 17:52 hwmon11
```
I used `hwmon*` in path, and in hddfancontrol v1 it works. Like using command `cd /sys/devices/platform/it87.2608/hwmon/hwmon*/` in bash.

desbma, 2025-01-07:
> I used hwmon* in path, and in hddfancontrol v1 it works. Like using command cd /sys/devices/platform/it87.2608/hwmon/hwmon*/ in bash.

In the log above, the literal string with `*` is passed to hddfancontrol. So you not started it from bash.
`*` interpolation is done by shells, not hddfancontrol, this has not changed since v1.

chbgdn, 2025-01-07:
Sure, I started it from openrc service. But v1 I start from openrc service too, and it works with `*` in path. So something in path processing between python and rust different I think.

desbma, 2025-01-07:
No, double check your service code or whatever you were doing with v1, `*` interpolation was **never** done by hddfancontrol, in v1 or v2, and never will.

If I had to guess, I' say you have probably quoted a string that was not quoted before in your service file (which are just shell scripts), or something like that.

Anyway I'm closing this, since the issue is not caused by hddfancontrol.

desbma, 2025-01-07:
See also: #51 #11

closed 2025-01-07 by desbma

chbgdn, 2025-01-07:
Yeah, thats same problem as mine, thanks. But I checked right now, and here results v1.6.2:
```
nas:~# hddfancontrol -d /dev/sda -p /sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3 --pwm-start-value 80 --pwm-stop-value 0
2025-01-07 22:58:23,958 WARNING [Main] sched_getscheduler is not supported on this system, leaving scheduler as is
2025-01-07 22:58:23,958 INFO [Main] Process reniced from 0 to -19
2025-01-07 22:58:24,055 INFO [sda SSDPR-CX400-256-G2] Will probe temperature with method DRIVETEMP
2025-01-07 22:58:24,056 INFO [Main] Maximum device temperature: 48 °C
2025-01-07 22:58:24,056 INFO [Fan #1] Setting fan speed to 90%
```
And here v2.0.0.b4:
```
nas:~# hddfancontrol daemon -d /dev/sda -p /sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3:80:0 -i 30s
INFO  [hddfancontrol::probe] Using probing method 'native Linux drivetemp' for drive 'sda SSDPR-CX400-256-G2'
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 10ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 15ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 22.5ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 33.75ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 50.625ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 75.9375ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 113.90625ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 170.859375ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 256.289062ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 384.433593ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 576.650389ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 864.975583ms
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 1s
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 1s
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 1s
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 1s
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 1s
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 1s
WARN  [hddfancontrol::pwm] "/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3" does not exist, retrying in 1s
Error: No such file or directory (os error 2)
```
So thats not scripting problem, etc. Just different path processing between python and rust.

desbma, 2025-01-07:
Nope, nothing to with Python and Rust, issue is 100% in your service file, and like I said, due to shell expansion.

In your first log, `/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3` is expanded by the shell because it is not quoted.
In your second log, `/sys/devices/platform/it87.2608/hwmon/hwmon*/pwm3:80:0` is not quoted either in the invocation, but **not** expanded because of the `:80:0` suffix (there is not file named `pwm3:80:0` in any `hwmonX` directory).

chbgdn, 2025-01-07:
Yeah, you right. So that's related to change pwm start/stop values setting (because now it given together with pwm path). Thanks for assistance.