desbma/hddfancontrol #66: pwm-test ../hwmon4/pwm1 No such file or directory (os error 2)

L3nny5, 2025-02-09:
Hey,

I'm trying to get hddfancontrol to work on my qnap NAS with True NAS OS. To control the fans I needed to install a module [qnap8528](https://github.com/0xGiddi/qnap8528)

I can control the fans with i.e. 
`root@nas:/sys/class/hwmon/hwmon4# echo 100 > pwm1`

However if I try to execute hddfanctontrol pwm-test an error occurs:
```
root@nas:/# hddfancontrol pwm-test --pwm /sys/class/hwmon/hwmon4/pwm1 
Error: No such file or directory (os error 2)
```

For me it seems that something which is normally there isn't. 
Getting the module to work I also had to use a workaround built in by the dev (skip_hw_check=true). 
With this option the module is not checking if the hardware is really present but force loads it.

At the moment I'm controling the fans via /usr/sbin/fancontrol which works fine. However this is not a persistent solution, as TrueNAS updates will overwrite such custom configs. So I want to use hddfancontrol via docker to not have issues with updates.

This is my fancontrol config file, which I use atm:
```
# Configuration file generated by pwmconfig, changes will be lost
INTERVAL=5
DEVPATH=hwmon3=devices/platform/coretemp.0 hwmon4=devices/platform/qnap8528
DEVNAME=hwmon3=coretemp hwmon4=qnap8528
FCTEMPS=hwmon4/pwm1=hwmon3/temp1_input
FCFANS= hwmon4/pwm1=hwmon4/fan3_input hwmon4/pwm1=hwmon4/fan2_input hwmon4/pwm1=hwmon4/fan1_input
MINTEMP=hwmon4/pwm1=40
MAXTEMP=hwmon4/pwm1=60
MINSTART=hwmon4/pwm1=22
MINSTOP=hwmon4/pwm1=165
MINPWM=hwmon4/pwm1=165
AVERAGE=hwmon4/pwm1=3
```

Any idea what I can do?

desbma, 2025-02-10:
What is the output of `ls -l /sys/class/hwmon/hwmon4`?

L3nny5, 2025-02-12:
```
root@server:~# ls -l /sys/class/hwmon/hwmon4
lrwxrwxrwx 1 root root 0 Feb  4 10:10 /sys/class/hwmon/hwmon4 -> ../../devices/platform/qnap8528/hwmon/hwmon4
```

```
root@server:~# hddfancontrol pwm-test --pwm /sys/devices/platform/qnap8528/hwmon/hwmon4/pwm1 
Error: No such file or directory (os error 2)
```

desbma, 2025-02-13:
Sorry, I need the output of `ls -l /sys/class/hwmon/hwmon4/` (with a `/` at the end).

To get more info you can try running `hddfancontrol` with `-v debug` arguments, and `RUST_BACKTRACE=1` environment variable set.

Also, if you can compile and run the code from master branch, I have added a bit more info when an error occurs.

desbma, 2025-02-14:
From the ls output I can see that the `/sys/devices/platform/f71882fg.656/pwm1_enable` file is missing, which causes the error. I didn't know this was possible. I'll change the code to support this.

> What exactly do you mean with environment variable. Do you mean my bash profile, where I set environment variables for my shell?

If you start `hddfancontrol` manually from your shell, yes. Otherwise that depends on how you start it. For example, systemd has [`Environment=`](https://www.freedesktop.org/software/systemd/man/latest/systemd.exec.html#Environment=) option for this.

desbma, 2025-02-14:
It should work with 1879b7070d62785e3dda8aaf78a9e208abed3acc, you can test by [compiling from sources](https://github.com/desbma/hddfancontrol?tab=readme-ov-file#from-source).

L3nny5, 2025-02-14:
I can confirm that compiling from source solved my issue. Thank you for your help!

```
root@server:/mnt/services01/fancontrol# hddfancontrol pwm-test --pwm /sys/class/hwmon/hwmon4/pwm1 
INFO  [hddfancontrol::fan] Running tests for 3 candidates to find RPM file correlated with PWM qnap8528/1, this may take some time
INFO  [hddfancontrol::fan] RPM file for qnap8528/1 resolved to "/sys/devices/platform/qnap8528/hwmon/hwmon4/fan3_input"
INFO  [hddfancontrol] Testing fan qnap8528/1, this may take a long time
INFO  [hddfancontrol::fan] Fan qnap8528/1 speed set to 100.0%
INFO  [hddfancontrol::fan] Fan qnap8528/1 speed set to 98.0%
INFO  [hddfancontrol::fan] Fan qnap8528/1 speed set to 96.1%
...
INFO  [hddfancontrol::fan] Fan qnap8528/1 speed set to 11.8%
INFO  [hddfancontrol] Fan qnap8528/1] start/stop thresholds: 30/20
```

closed 2025-02-14 by L3nny5