desbma/hddfancontrol #67: [Enhancement] Add support for scsi drives - HBAs

philipwilk, 2025-03-04:
From what I have seen, some people have reported that this works fine with sas drives for them. That is great. Mine do not. My NAS' drives are running through a HBA, and they are sas, so they are scsi drives.
`hdparm` cannot handle scsi drives - it retrieves no info about them. `sdparm` is the counterpart that is made for scsi drives, and it reads from them just fine.

<details>
<summary>hdparm:</summary>

[image]
</details>

<details>
<summary>sdparm:</summary>

[image]
</details>

This is an issue, as ``hdparm`` constantly thinks the drives are at standby, when they are not. ``sdparm`` does not have this issue. This results in hddfancontrol never adjusting the fan speed ever, resulting in the drives getting toasty (saw them get up to 53c!, fans barely spinning, clearly not working) - I manually set the minimum to 50% so they are at a reasonable temp, but this is not a solution as when they get under load and get hotter, or add other drives and the idle temp is higher, this will not suffice.
<details>
<summary>

`hddfancontrol.service`
</summary>

```log
Mar 04 16:01:40 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Drive sdc ST2000NM0001 is sleeping
Mar 04 16:01:40 itxserve hddfancontrol[68987]: INFO  [hddfancontrol] All drives are spun down
Mar 04 16:01:40 itxserve hddfancontrol[68987]: TRACE [hddfancontrol::fan] Fan nct6775.656/4 speed unchanged: 50.0%
Mar 04 16:01:40 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Will sleep at most 29.993924981s
Mar 04 16:02:10 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Drive sdb ST2000NM0001 state: standby
Mar 04 16:02:10 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Drive sdb ST2000NM0001 is sleeping
Mar 04 16:02:10 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Drive sdc ST2000NM0001 state: standby
Mar 04 16:02:10 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Drive sdc ST2000NM0001 is sleeping
Mar 04 16:02:10 itxserve hddfancontrol[68987]: INFO  [hddfancontrol] All drives are spun down
Mar 04 16:02:10 itxserve hddfancontrol[68987]: TRACE [hddfancontrol::fan] Fan nct6775.656/4 speed unchanged: 50.0%
Mar 04 16:02:10 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Will sleep at most 29.994184121s
Mar 04 16:02:40 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Drive sdb ST2000NM0001 state: standby
Mar 04 16:02:40 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Drive sdb ST2000NM0001 is sleeping
Mar 04 16:02:40 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Drive sdc ST2000NM0001 state: standby
Mar 04 16:02:40 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Drive sdc ST2000NM0001 is sleeping
Mar 04 16:02:40 itxserve hddfancontrol[68987]: INFO  [hddfancontrol] All drives are spun down
Mar 04 16:02:40 itxserve hddfancontrol[68987]: TRACE [hddfancontrol::fan] Fan nct6775.656/4 speed unchanged: 50.0%
Mar 04 16:02:40 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Will sleep at most 29.994223945s
Mar 04 16:03:10 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Drive sdb ST2000NM0001 state: standby
Mar 04 16:03:10 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Drive sdb ST2000NM0001 is sleeping
Mar 04 16:03:10 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Drive sdc ST2000NM0001 state: standby
Mar 04 16:03:10 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Drive sdc ST2000NM0001 is sleeping
Mar 04 16:03:10 itxserve hddfancontrol[68987]: INFO  [hddfancontrol] All drives are spun down
Mar 04 16:03:10 itxserve hddfancontrol[68987]: TRACE [hddfancontrol::fan] Fan nct6775.656/4 speed unchanged: 50.0%
Mar 04 16:03:10 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Will sleep at most 29.994104214s
Mar 04 16:03:40 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Drive sdb ST2000NM0001 state: standby
Mar 04 16:03:40 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Drive sdb ST2000NM0001 is sleeping
Mar 04 16:03:40 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Drive sdc ST2000NM0001 state: standby
Mar 04 16:03:40 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Drive sdc ST2000NM0001 is sleeping
Mar 04 16:03:40 itxserve hddfancontrol[68987]: INFO  [hddfancontrol] All drives are spun down
Mar 04 16:03:40 itxserve hddfancontrol[68987]: TRACE [hddfancontrol::fan] Fan nct6775.656/4 speed unchanged: 50.0%
Mar 04 16:03:40 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Will sleep at most 29.994151692s
Mar 04 16:04:10 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Drive sdb ST2000NM0001 state: standby
Mar 04 16:04:10 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Drive sdb ST2000NM0001 is sleeping
Mar 04 16:04:10 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Drive sdc ST2000NM0001 state: standby
Mar 04 16:04:10 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Drive sdc ST2000NM0001 is sleeping
Mar 04 16:04:10 itxserve hddfancontrol[68987]: INFO  [hddfancontrol] All drives are spun down
Mar 04 16:04:10 itxserve hddfancontrol[68987]: TRACE [hddfancontrol::fan] Fan nct6775.656/4 speed unchanged: 50.0%
Mar 04 16:04:10 itxserve hddfancontrol[68987]: DEBUG [hddfancontrol] Will sleep at most 29.994115651s
```
</details>

In ``src/main.rs``, when the drives are found to be at standby and sleep probing is unsupported, it assumes the max_drive_temp to be none, so it runs the fans at basically off.
<details>
<summary>

`src/main.rs:175`
</summary>

```rust
  let max_drive_temp = drive_probers
      .iter_mut()
      .zip(drives.iter())
      .map(|((prober, supports_probing_sleeping), drive)| {
          let state = drive
              .state()
              .with_context(|| format!("Failed to get drive {drive} state"))?;
          log::debug!("Drive {drive} state: {state}");
          let temp = if state.is_spun_down() && !*supports_probing_sleeping {
              log::debug!("Drive {drive} is sleeping");
              None
          } else {
              let temp = prober
                  .probe_temp()
                  .with_context(|| format!("Failed to get drive {drive} temp"))?;
              log::debug!("Drive {drive}: {temp}°C");
              Some(temp)
          };
          Ok(temp)
      })
      .collect::<anyhow::Result<Vec<_>>>()
      .context("Failed to get maximum drive temperature")?
      .into_iter()
      .flatten()
      .reduce(f64::max);
```
</details>

This could be resolved by making `state_()` automatically decide to use `hdparm` or `sdparm` depending on which is supported (it is one or the other). In this case, the `sdparm` equivalent of `hdparm -C /dev/path` to get the state is `sdparm --command=ready /dev/path`, which either returns `Ready` or `Not Ready`.
<details>
<summary>See</summary>

[image]
[image]
</details>

The drive maximum/trip/reference temperature can then be retrieved via `sg_logs` or `smartctl` (which afaik uses sg_logs so might as well).
<details>
<summary>Reference Temperature/Drive Trip Temperature</summary>

[image]
[image]

</details>

There is a neat table with the equivalents [here](https://doug-gilbert.github.io/sdparm.html)

Could trivially hotfix it by trying with `sdpam` if the `hdparm` response contains `"bad/missing sense data"`,  but a more robust solution would probably be to determine whether to use `hdparm` or `sdparm` when the drives are initially found on daemon start and then just calling the appropriate one directly. Something along the lines of this could work?

```
lsblk /dev/path -do tran:
switch
  sas -> sdparm
  ata | sata -> hdparm
  nvme -> nvme-cli
```

desbma, 2025-03-07:
Yes it makes sense to use `sdparm` for SAS drives. The main issue is I don't have such drives, so I would need you to confirm it works.

Can you post the output (as text), of:
* `hdparm -C /dev/XXX && echo $?` where `/dev/XXX` is a SAS drive
* `sdparm --command=ready /dev/XXX` where `/dev/XXX` is an active SAS drive
* `sdparm --command=ready /dev/XXX` where `/dev/XXX` is an spun down SAS drive

I have added a small fix in the [`sas`](https://github.com/desbma/hddfancontrol/tree/sas) branch, it should error out immediately in your case instead of assuming the drives are in standby state.

philipwilk, 2025-03-08:
```
[philip@itxserve:~]$ sudo hdparm -C /dev/sda && echo $?

/dev/sda:
SG_IO: bad/missing sense data, sb[]:  70 00 05 00 00 00 00 0a 00 00 00 00 20 00 01 cf 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 drive state is:  standby
0

[philip@itxserve:~]$ sudo sdparm --command=ready /dev/sda
    /dev/sda: SEAGATE   ST2000NM0001      0002
Ready

[philip@itxserve:~]$ sudo sdparm --flexible --command=stop /dev/sda
    /dev/sda: SEAGATE   ST2000NM0001      0002

[philip@itxserve:~]$ sudo sdparm --command=ready /dev/sda
    /dev/sda: SEAGATE   ST2000NM0001      0002
Not ready
```

desbma, 2025-03-08:
Can you test the [`sas` branch](https://github.com/desbma/hddfancontrol/tree/sas) ? 

State probing should now fallback to `sdparm` for SAS drives.

However I am not sure temperature probing will then work, so please test.

desbma, 2025-03-22:
@philipwilk 
Were you able to test the code in the [`sas` branch](https://github.com/desbma/hddfancontrol/tree/sas)?
If you are unable to build from source, I can provide you a static binary or a Debian package.

philipwilk, 2025-03-22:
Sorry, got busy with work, will take a look tomorrow.

philipwilk, 2025-03-23:
new log:

```
Mar 23 19:17:38 itxserve hddfancontrol[18627]: TRACE [hddfancontrol::device::drive] hdparm -I /dev/sda
Mar 23 19:17:38 itxserve hddfancontrol[18627]: TRACE [hddfancontrol::device::drive] smartctl -i /dev/sda
Mar 23 19:17:38 itxserve hddfancontrol[18627]: DEBUG [hddfancontrol::device::drive] sda ST2000NM0001: Will use sdparm state probing method
Mar 23 19:17:38 itxserve hddfancontrol[18627]: TRACE [hddfancontrol::device::drive] hdparm -I /dev/sde
Mar 23 19:17:38 itxserve hddfancontrol[18627]: TRACE [hddfancontrol::device::drive] smartctl -i /dev/sde
Mar 23 19:17:38 itxserve hddfancontrol[18627]: DEBUG [hddfancontrol::device::drive] sde ST2000NM0001: Will use sdparm state probing method
Mar 23 19:17:38 itxserve hddfancontrol[18627]: INFO  [hddfancontrol::probe] Drive 'sda ST2000NM0001' does not support probing method 'native Linux drivetemp': "/sys/block/sda/../../hwmon" does not exist
Mar 23 19:17:38 itxserve hddfancontrol[18627]: INFO  [hddfancontrol::probe] Drive 'sda ST2000NM0001' does not support probing method 'hdparm Hitachi/HGST': hdparm returned soft error
Mar 23 19:17:39 itxserve hddfancontrol[18627]: INFO  [hddfancontrol::probe] Drive 'sda ST2000NM0001' does not support probing method 'smartctl SCT': Failed to parse smartctl SCT temp output
Mar 23 19:17:39 itxserve hddfancontrol[18627]: INFO  [hddfancontrol::probe] Using probing method 'hddtemp daemon' for drive 'sda ST2000NM0001'
Mar 23 19:17:39 itxserve hddfancontrol[18627]: INFO  [hddfancontrol::probe] Drive 'sde ST2000NM0001' does not support probing method 'native Linux drivetemp': "/sys/block/sde/../../hwmon" does not exist
Mar 23 19:17:39 itxserve hddfancontrol[18627]: INFO  [hddfancontrol::probe] Drive 'sde ST2000NM0001' does not support probing method 'hdparm Hitachi/HGST': hdparm returned soft error
Mar 23 19:17:39 itxserve hddfancontrol[18627]: INFO  [hddfancontrol::probe] Drive 'sde ST2000NM0001' does not support probing method 'smartctl SCT': Failed to parse smartctl SCT temp output
Mar 23 19:17:39 itxserve hddfancontrol[18627]: INFO  [hddfancontrol::probe] Using probing method 'hddtemp daemon' for drive 'sde ST2000NM0001'
Mar 23 19:17:39 itxserve hddfancontrol[18627]: DEBUG [hddfancontrol] Drive sda ST2000NM0001 state: active/idle
Mar 23 19:17:39 itxserve hddfancontrol[18627]: DEBUG [hddfancontrol] Drive sda ST2000NM0001: 39°C
Mar 23 19:17:39 itxserve hddfancontrol[18627]: DEBUG [hddfancontrol] Drive sde ST2000NM0001 state: active/idle
Mar 23 19:17:39 itxserve hddfancontrol[18627]: DEBUG [hddfancontrol] Drive sde ST2000NM0001: 37°C
Mar 23 19:17:39 itxserve hddfancontrol[18627]: INFO  [hddfancontrol] Max drive temperature: 39°C
Mar 23 19:17:39 itxserve hddfancontrol[18627]: INFO  [hddfancontrol::fan] PWM nct6775.656/4 mode set from OFF to SOFTWARE
Mar 23 19:17:39 itxserve hddfancontrol[18627]: TRACE [hddfancontrol::pwm] Set PWM nct6775.656/4 to 132
Mar 23 19:17:39 itxserve hddfancontrol[18627]: INFO  [hddfancontrol::fan] Fan nct6775.656/4 speed set to 50.0%
Mar 23 19:17:39 itxserve hddfancontrol[18627]: DEBUG [hddfancontrol] Will sleep at most 29.996679633s
Mar 23 19:18:09 itxserve hddfancontrol[18627]: DEBUG [hddfancontrol] Drive sda ST2000NM0001 state: active/idle
Mar 23 19:18:09 itxserve hddfancontrol[18627]: DEBUG [hddfancontrol] Drive sda ST2000NM0001: 39°C
Mar 23 19:18:09 itxserve hddfancontrol[18627]: DEBUG [hddfancontrol] Drive sde ST2000NM0001 state: active/idle
Mar 23 19:18:09 itxserve hddfancontrol[18627]: DEBUG [hddfancontrol] Drive sde ST2000NM0001: 37°C
Mar 23 19:18:09 itxserve hddfancontrol[18627]: INFO  [hddfancontrol] Max drive temperature: 39°C
Mar 23 19:18:09 itxserve hddfancontrol[18627]: TRACE [hddfancontrol::fan] Fan nct6775.656/4 speed unchanged: 50.0%
Mar 23 19:18:09 itxserve hddfancontrol[18627]: DEBUG [hddfancontrol] Will sleep at most 29.996086113s
```

philipwilk, 2025-03-23:
Live temp reading looks fine, can't see if it is reading the max/trip temp correctly, might be worth putting a ``log::debug``/``log::trace`` into `src/device/hwmon.rs` so I can verify that?

desbma, 2025-03-23:
Your log to me indicates that hddfancontrol is working fine. I will merge this in master branch then. Thanks for testing and reporting back.

The code for hwmon is unrelated to hard drives, it is used when passing the `--hwmon` command line parameter, typically to regulate fan speed from an additional CPU or motherboard temp probe.

closed 2025-03-23 by desbma

philipwilk, 2025-03-23:
Which tool does hddfancontrol use to get the trip temp? I'd just like to check manually to reassure myself its got the right number.

desbma, 2025-03-23:
It does not try to guess the temperature limits, you set them from command line.

From `man hddfancontrol` or `man hddfancontrol-daemon`:
```
  -t, --drive-temp-range <TEMP> <TEMP>
          Temperatures in Celcius at which the fan(s) will be set to minimum/maximum speed [default: 30 50]
```

KALRONG, 2025-04-04:
Sorry for bumping on this closed issue just noting that this requires sdparm to be installed and this is missing on the installation instructions (hdparm is specified) other than that it works like a charm thanks a lot! ^^

desbma, 2025-04-04:
@KALRONG added in ad353f215e20c65a1dc0f7f812b8cf46fae3be7a