desbma/hddfancontrol #68: More granular controle on fan speed based on temp with custom fan curves

Breigner01, 2025-03-22:
It would be awesome to be able to set a custom fan curve based on drive temp.

I would really love to as I have some Noctua industrial fans for my drive compartment on a fractal design node 804.

And the fans are either at 25% or 30%, nothing in between and I would love to be able to make a much smoother curve before 45°C so that it doesn't sound like a plane taking off unless temp is too high.

I totally understand that the default behavior may except some standard fans that don't go over 1500 rpm but it would be great to allow customizing the curve for those with 2000+ rpm fans (3000rpm in my case).

If there is anything I can do to help implement it, please let me know, I would gladly take time to help implement and test that.

Breigner01, 2025-03-23:
I think I haven't phrased it the best.

What I mean is that hddfancontrol was probably built with standard case fans in mind, which don't go as high in terms of rpm so a linear fan curves works better with those in terms of noise/cooling ratio.

I currently use 30˚C as min temp and 50˚C as max temp.

So 25% would mean the highest temp is 35˚C whereas 30% would mean 36˚C.

If I'm doing the math right in my head, lowering the min temp would slightly increase the percentage for the same temperature.

And I wanted a custom fan curve to have more granular control and make it less noisy at lower temps and boost really fast in case of high temps, I wanted to have something like 0-70% for 30-50˚C and over 50˚C boost to probably 100%

desbma, 2025-03-25:
> What I mean is that hddfancontrol was probably built with standard case fans in mind, which don't go as high in terms of rpm so a linear fan curves works better with those in terms of noise/cooling ratio.

That is not the case (pun unintended). But I'm still failing to see how high RPM fans are related to the issue.

> make it less noisy at lower temps and boost really fast in case of high temps

Currently hddfancontrol lets you build a linear curve from 3 variables : min speed, temp at which speed start to increases, and temp at which fans blow at full speed.

So I think you can already achieve exactly what you describe by:
* setting a minimum speed (`--min-fan-speed-prct`) to a silent level
* setting a quite high minimum temperature, say 45°C
* setting 50°C as max temperature

Breigner01, 2025-03-31:
Hi,

Thank you for the help.

I will try this.

I understand that hddfancontrol was not meant to be used with full on fan curves and will look into other solutions.

Again, thank you for your time.

closed 2025-03-31 by Breigner01