desbma/hddfancontrol #71: WD DC HC550 SAS Drive "No probing method found for drive"

h0m3us3r, 2025-07-27:
No existing method supports WD DC HC550 SAS drive (except maybe hddtemp which I didn't try)

```
INFO  [hddfancontrol::probe] Drive 'sda WUH721818AL5200' does not support probing method 'native Linux drivetemp': "/sys/block/sda/../../hwmon" does not exist
INFO  [hddfancontrol::probe] Drive 'sda WUH721818AL5200' does not support probing method 'hdparm Hitachi/HGST': hdparm failed with code exit status: 5
INFO  [hddfancontrol::probe] Drive 'sda WUH721818AL5200' does not support probing method 'smartctl SCT': Failed to parse smartctl SCT temp output
INFO  [hddfancontrol::probe] Drive 'sda WUH721818AL5200' does not support probing method 'hddtemp daemon': Connection refused (os error 111)
INFO  [hddfancontrol::probe] Drive 'sda WUH721818AL5200' does not support probing method 'hddtemp invocation': No such file or directory (os error 2)
INFO  [hddfancontrol::probe] Drive 'sda WUH721818AL5200' does not support probing method 'smartctl SMART attribute': Failed to parse smartctl attribute output, or no temp attribute
Error: Failed to setup drive probers

Caused by:
    No probing method found for drive "/dev/sda"

```

megatog615, 2025-07-27:
There's supposed to be sdparm support but, like in your case, hddfancontrol doesn't support my SAS drives if I use `-d scsi`.

desbma, 2025-07-29:
@h0m3us3r can you post the output of `smartctl -a` for the drive in question? Your PR #72 looks good but I want to be sure we can't get temperature with current methods before adding a new dependency.
You message in https://github.com/AnalogJ/scrutiny/issues/788 seem to indicate `smartctl` can get temperature?

megatog615, 2025-07-29:
I'm not h0m3us3r but my drives seem to have the same problem. smartctl -a shows a drive temperature but hddfancontrol doesn't see it for some reason.

```
smartctl 7.3 2022-02-28 r5338 [x86_64-linux-6.1.0-37-amd64] (local build)
Copyright (C) 2002-22, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Vendor:               HGST
Product:              HUS726040AL5210
Revision:             A907
Compliance:           SPC-4
User Capacity:        4,000,787,030,016 bytes [4.00 TB]
Logical block size:   512 bytes
Physical block size:  4096 bytes
LU is fully provisioned
Rotation Rate:        7200 rpm
Form Factor:          3.5 inches
Logical Unit id:      0x5000cca244228a8c
Serial number:        N8GLZWLZ
Device type:          disk
Transport protocol:   SAS (SPL-4)
Local Time is:        Tue Jul 29 16:10:01 2025 EDT
SMART support is:     Available - device has SMART capability.
SMART support is:     Enabled
Temperature Warning:  Enabled

=== START OF READ SMART DATA SECTION ===
SMART Health Status: OK

Current Drive Temperature:     42 C
Drive Trip Temperature:        85 C

Accumulated power on time, hours:minutes 49198:27
Manufactured in week 40 of year 2016
Specified cycle count over device lifetime:  50000
Accumulated start-stop cycles:  20
Specified load-unload count over device lifetime:  600000
Accumulated load-unload cycles:  1001
Elements in grown defect list: 12

Vendor (Seagate Cache) information
  Blocks sent to initiator = 61266541672398848

Error counter log:
           Errors Corrected by           Total   Correction     Gigabytes    Total
               ECC          rereads/    errors   algorithm      processed    uncorrected
           fast | delayed   rewrites  corrected  invocations   [10^9 bytes]  errors
read:          0        3         0         3    3245020      40924.965           0
write:         0    22031         0     22031   51942851     425353.422           0
verify:        0       24         0        24     892865      12807.604           0

Non-medium error count:        0

SMART Self-test log
Num  Test              Status                 segment  LifeTime  LBA_first_err [SK ASC ASQ]
     Description                              number   (hours)
# 1  Background short  Completed                   -   49184                 - [-   -    -]
# 2  Background long   Completed                   -   49172                 - [-   -    -]
# 3  Background short  Completed                   -   49160                 - [-   -    -]
# 4  Background short  Completed                   -   49136                 - [-   -    -]
# 5  Background short  Completed                   -   49112                 - [-   -    -]
# 6  Background short  Completed                   -   49088                 - [-   -    -]
# 7  Background short  Completed                   -   49064                 - [-   -    -]
# 8  Background short  Completed                   -   49040                 - [-   -    -]
# 9  Background short  Completed                   -   49016                 - [-   -    -]
#10  Background long   Completed                   -   49004                 - [-   -    -]
#11  Background short  Completed                   -   48992                 - [-   -    -]
#12  Background short  Completed                   -   48968                 - [-   -    -]
#13  Background short  Completed                   -   48944                 - [-   -    -]
#14  Background short  Completed                   -   48920                 - [-   -    -]
#15  Background short  Completed                   -   48896                 - [-   -    -]
#16  Background short  Completed                   -   48872                 - [-   -    -]
#17  Background short  Completed                   -   48848                 - [-   -    -]
#18  Background long   Completed                   -   48836                 - [-   -    -]
#19  Background short  Completed                   -   48824                 - [-   -    -]
#20  Background short  Completed                   -   48800                 - [-   -    -]

Long (extended) Self-test duration: 34237 seconds [9.5 hours]
```

desbma, 2025-07-29:
It should be fixed by 3ba1da1179e47adb0f4ea892070c2f06986b457a, can you please test and confirm?

megatog615, 2025-07-29:
I hate to do this but I do not want to install the rust toolchain(s) to get this going. I am running debian stable on my NAS and do not want to use rustup.

desbma, 2025-07-29:
Here is a Debian package made from commit aa425e9fb5fdb3d0d6bf7de52618182fc6c01890: https://x0.at/Mhci.deb

closed 2025-07-30 by desbma